PX4/PX4-Autopilot #9156: SITL Simulation of iris_rplidar using Ray class does not work with Laser Altimeter

lolster, 2018-03-25:
We have some issues trying to run the Gazebo simulation of the `iris_rplidar` model using the CPU version (`Ray` class).  
We are using Gazebo 7, ROS Kinetic and Ubuntu 16.04. The simulation is running with PX4 v1.7.3.

The simulation seems to work fine as is from the repo using the `GpuRay` version. However when we switch it to the CPU version when trying to run the simulation using the command `roslaunch px4 mavros_posix_sitl.launch vehicle:=iris_opt_flow` it fails with a SegFault (see below).

```
...
[INFO] [1521963013.283246, 0.000000]: Loading model XML from file
[INFO] [1521963013.283508, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1521963014.380615270]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1521963014.381053625]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[INFO] [1521963015.696928, 0.000000]: Calling service /gazebo/spawn_sdf_model
[INFO] [1521963016.466065, 0.002000]: Spawn status: SpawnModel: Successfully spawned entity
[vehicle_spawn_sushrith_hp_18072_8143607915859286538-5] process has finished cleanly
log file: /home/sushrith/.ros/log/5a1e3f4e-2ffe-11e8-b2b4-142d276b1a25/vehicle_spawn_sushrith_hp_18072_8143607915859286538-5*.log
Segmentation fault (core dumped)
[gazebo-3] process has died [pid 18128, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -e ode /home/sushrith/git/Firmware/Tools/sitl_gazebo/worlds/empty.world __name:=gazebo __log:=/home/sushrith/.ros/log/5a1e3f4e-2ffe-11e8-b2b4-142d276b1a25/gazebo-3.log].
log file: /home/sushrith/.ros/log/5a1e3f4e-2ffe-11e8-b2b4-142d276b1a25/gazebo-3*.log
^C[mavros-6] killing on exit
[gazebo_gui-4] killing on exit
[sitl-2] killing on exit

Exiting...
pxh> Shutting down
Restoring terminal
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

And the sdf file for the rplidar model with changes to use the `Ray` class (CPU).

```xml
<?xml version="1.0" ?>
<sdf version="1.5">
  <model name="rplidar">
    <link name="link">

      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>0.19</mass>
        <inertia>
          <ixx>4.15e-6</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>2.407e-6</iyy>
          <iyz>0</iyz>
          <izz>2.407e-6</izz>
        </inertia>
      </inertial>

      <visual name="visual">
        <geometry>
          <box>
            <size>0.02 0.05 0.05</size>
          </box>
        </geometry>
      </visual>

      <sensor name="laser" type="ray">
        <ray>
          <scan>
            <horizontal>
              <samples>360</samples>
              <resolution>1</resolution>
              <min_angle>-3.14</min_angle>
              <max_angle>3.14</max_angle>
            </horizontal>
          </scan>
          <range>
            <min>0.2</min>
            <max>6</max>
            <resolution>0.05</resolution>
          </range>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.01</stddev>
          </noise>
        </ray>
        <plugin name="laser" filename="libRayPlugin.so" />
        <plugin name="gazebo_ros_head_rplidar_controller" filename="libgazebo_ros_laser.so">
          <topicName>laser/scan</topicName>
          <frameName>rplidar_link</frameName>
        </plugin>
        <always_on>1</always_on>
        <update_rate>5.5</update_rate>
        <visualize>true</visualize>
      </sensor>
    </link>
  </model>
</sdf>

<!-- vim: set et fenc= ff=unix sts=0 sw=2 ts=2 : -->
```

Following the suggestions in #6755 we removed the Laser altimeter after which the simulation ran through just fine and seemed to give coherent values.

Any ideas as to what might be wrong?

Working here with @anushkumar27.

mhkabir, 2018-03-25:
The laser altimeter and the rplidar have some weird conflict when used simultaneously. This is most likely a Gazebo bug.

sarahToonsi, 2018-12-19:
> The laser altimeter and the rplidar have some weird conflict when used simultaneously. This is most likely a Gazebo bug.

can you please explain what do you mean by a laser altimeter cause I cant find that in the SDF file of the iris or the rplidar

closed 2019-04-09 by stale[bot]