PX4/PX4-Autopilot #9160: Problem with internal magnetic sensors PIXHAWK 3 PRO

CediBosshard, 2018-03-26:
**Firmware v1.7.3 & QGC v3.3.0**
During test flights I noticed that the drone does not fly the route I specified, but always flies in any other direction. Several times I recalibrated all sensor systems with QGroundControl and still got the following warining: _preflight failed mag sensors inconsistent_

I have read the following forum posts:
- preflight failed mag sensors inconsistent #8475
- Solo PREFLIGHT FAIL: MAG SENSORS INCONSISTENT #8305
- preflight failed : mag sensors inconsistent #24
- Pixracer magnetometer preflight error #8098

After a few searches, I found the cause:
One internal magnetic sensor provided data that differed greatly from the other internal magnetic sensor and the magnetic sensor of the external [GPS compass module](https://drotek.com/shop/en/u-blox/876-module-gps-ublox-neo-m8n-magnetometre-lis3mdl-pixhawk-3-pro.html). I created plots in all three axes using FlightPlot.

Internal magnetic sensor (sensor_mag_0 & sensor_mag_1):
- [MPU-9250](https://www.invensense.com/products/motion-tracking/9-axis/mpu-9250/)
- [LIS3MDL](https://www.digikey.com/catalog/en/partgroup/lis3mdl/41283)

External magnetic sensor (sensor_mag_2):
- [LIS3MDL](https://www.digikey.com/catalog/en/partgroup/lis3mdl/41283)

I can’t tell which sensor is exactly mag_0 and which is mag_1.

[image]
[image]
[image]

In the X-axis the magnetic sensor 1 has an offset, in the Y- & Z-axis it looks as if the signal is mirrored. 

First tried to change the orientation of the sensor (CAL_MAG1_ROT), but after the QGC after the fifth time, set the value back to _Internal mag_, we left this.
As a result, I have deactivated the magnetic sensor in QGroundControl, which has a systematic deviation (CAL_MAG1_EN set to _Disabled_).

[image]

Since that moment, the drone has been flying exactly as planned! 

My questions are this now:
- Where could the error come from with the wrong signal?  
- How can you fix it yourself?

Thanks for your help!
Regards Cédi

xdwgood, 2018-04-03:
Just personal advice：　you can set `CAL_MAG2_ROT` = Internal mag

CediBosshard, 2018-04-05:
What is the point if I set CAL_MAG2_ROT = Internal?

closed 2019-07-08 by stale[bot]