PX4/PX4-Autopilot #9165: Wrong note

BazookaJoe1900, 2018-03-26:
I think that the next note is wrong:
https://github.com/PX4/Firmware/blob/1b174eeca220a1176f2c3146515a4ccacf592e76/src/drivers/imu/mpu9250/mpu9250.cpp#L464

the LPF is of 41Hz, and not 20Hz.
https://github.com/PX4/Firmware/blob/cf54023c9613a4f5b783ccd2d2f7c7e0adde968f/src/drivers/imu/mpu9250/mpu9250.h#L191

Am I correct?
Do you see issue of chnaging it to 20Hz because I think that on my drone the props freq is in about 35Hz and that mean that the LPF doesn't works for me....

BazookaJoe1900, 2018-12-14:
anyone?

hamishwillee, 2018-12-17:
@bkueng Assigned to you to find someone to comment on this :-)

bkueng, 2019-01-07:
Yes that note is wrong, and it's already fixed on master: https://github.com/PX4/Firmware/blob/master/src/drivers/imu/mpu9250/mpu9250.cpp#L490.
BTW it's now 92 by default, and you can use `IMU_GYRO_CUTOFF` and `IMU_ACCEL_CUTOFF` to change the cutoff frequencies of the software filters.

closed 2019-01-07 by bkueng