PX4/PX4-Autopilot #9170: Bottle_drop not working in v1.5.4

myeabhishek, 2018-03-27:
Hi Team,

I am not able to make bottle_drop work in v1.5.4. I think the issue is with setting the mixer file. Can someone please guide me on how to set the mixer in this particular version? 

I know how to set the mixer with newer version but unfortunately that approach isn't working.

LorenzMeier, 2018-03-27:
Support for the old outback challenge code has been phased out. This is not an issue but an informed decision of the maintenance team.

closed 2018-03-27 by LorenzMeier

hamishwillee, 2018-03-27:
@LorenzMeier Should the source then be removed from the tree? https://github.com/PX4/Firmware/tree/master/src/examples/bottle_drop

myeabhishek, 2018-03-28:
I am not able to set the mixer for actuator_control_group2 in v1.5.4. I am able to get it to work in newer versions. Can someone help me with that?