PX4/PX4-Autopilot #9178: NO position lock in SITL

ChenYunan, 2018-03-28:
**Bug Report** 

I updated my gazebo from 8 to 10. And make the SITL: make posix gazebo, the iris denied takeoff.

**pxh>** commander takeoff
WARN  [commander] rejecting takeoff, no position lock yet. Please retry..

**pxh>** commander check
WARN  [commander_tests] PREFLIGHT FAIL: GPS RECEIVER MISSING
WARN  [commander_tests] PREFLIGHT FAIL: EKF NOT USING GPS
WARN  [commander] Preflight check: FAILED
WARN  [commander] Prearm check: OK
**pxh>** commander_tests
INFO  [commander_tests] RUNNING TEST: armingStateTransitionTest
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_INIT - ARMING_STATE_ARMED
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_STANDBY - ARMING_STATE_ARMED_ERROR
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_ARMED - ARMING_STATE_INIT
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_ARMED - ARMING_STATE_REBOOT
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_ARMED_ERROR - ARMING_STATE_ARMED
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_ARMED_ERROR - ARMING_STATE_REBOOT
WARN  [commander_tests] Preflight check resolved, reboot before arming
WARN  [commander_tests] Preflight check resolved, reboot to complete
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_REBOOT - ARMING_STATE_ARMED
WARN  [commander_tests] TRANSITION_DENIED: ARMING_STATE_IN_AIR_RESTORE - ARMING_STATE_STANDBY
WARN  [commander_tests] NOT ARMING: Press safety switch first!
INFO  [commander_tests] TEST PASSED: armingStateTransitionTest
INFO  [commander_tests] RUNNING TEST: mainStateTransitionTest
INFO  [commander_tests] TEST PASSED: mainStateTransitionTest
INFO  [commander_tests] RUNNING TEST: isSafeTest
INFO  [commander_tests] TEST PASSED: isSafeTest
INFO  [commander_tests] ALL TESTS PASSED
INFO  [commander_tests]   Tests passed :      3
INFO  [commander_tests]   Tests failed :      0
INFO  [commander_tests]   Tested assertions : 167

How can I fix this issue?
Thank you.

hamishwillee, 2018-03-28:
Gazebo 10 is AFAIK not supported. Gazebo 9 support went in today I think. @julianoes  might have tried this.

ChenYunan, 2018-03-28:
@hamishwillee Thanks for your reply! However, I tried on Gazebo 9 and Gazebo 8 (using apt-get install/autoremove), it had the same problem.
I can start 'make posix gazebo' successfully before I updated the version of Gazebo from 8 to 10.
But this time, I don't know what's going on. I think maybe it's not about the Gazebo version. Instead, there is no information (come from 'make posix jmavsim') below:
**'INFO  [lib__ecl] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
 INFO  [lib__ecl] EKF GPS checks passed (WGS-84 origin set)
 INFO  [lib__ecl] EKF commencing GPS fusion
 INFO  [tone_alarm] home_set'**
[image]

ChenYunan, 2018-04-04:
However, it is able to work with lpe instead of ekf2.

julianoes, 2018-04-04:
@ChenYunan we only added compatibility with Gazebo 9, not 10 yet. If you can find out what the issue is and make a pull request that would be great.

ChenYunan, 2018-04-23:
I think it has nothing to do with the Gazebo version. I re-clone and build the px4 Firmware with tag 1.7.4, and running SITL with Gazebo 9,  the issue is gone. This Q&A can be closed. Thank you!

closed 2018-04-23 by dagar