PX4/PX4-Autopilot #9180: IMU and Gyroscope Calibration Parameters

SubMishMar, 2018-03-28:
I calibrated my Compass, Gyroscope and Accelerometer for a pixhawk with px4 firmware. I am doing visual inertial slam so I need to know the IMU and Gyro Calibration parameters but I don't know where to find those. Will be glad if someone helps me here.

xdwgood, 2018-04-03:
@SubMishMar  you need connect to QGC , and switch params set. you will find these calibration.
gyro: `CAL_GYRO0_XOFF` `CAL_GYRO0_YOFF` `CAL_GYRO0_ZOFF` `CAL_GYRO0_XSCALE ``CAL_GYRO0_YSCALE` `CAL_GYRO0_ZSCALE`
acc : `CAL_ACC0_XOFF`.... `CAL_ACC0_XSCALE`...(X/Y/Z)
mag : `CAL_MAG0_XOFF` `CAL_MAG0_XSCALE` ...(X/Y/Z) :smiley:

xdwgood, 2018-04-03:
you aslo can upload your log , You can find these parameters on the log loaded website (below the site)
 link: https://review.px4.io/plot_app

closed 2019-01-27 by mhkabir