PX4/PX4-Autopilot #9184: Error "Mag 0 fail: STALE!" in Pixhawk2.1 + Here GNSS with PX4 v1.7.3

bigbellmercy, 2018-03-29:
What does this error means?

Error message: "Mag 0 fail: STALE!"

Environment:
Pixhawk2.1(px4fmu-v3_default) + Here GNSS with PX4 v1.7.3

Currently, the magnetometer inside of Here GNSS is not used in PX4 v1.7.3. So this happens?

Related issues:
#7398 and #7395

dagar, 2018-03-29:
If you can share a log (https://logs.px4.io/) or reproduce it and do some debugging we might be able to determine more.

bigbellmercy, 2018-03-30:
Thanks.
Here is the [log](https://logs.px4.io/plot_app?log=6e5808c7-2679-49e7-88e7-9c5d2a05d6fa).

I think the problem is reproducible because my Pixhawk is new one and the PX4 firmware had no change except the build bug fix from 'external' to 'externalL' variable names.

closed 2018-05-10 by PX4BuildBot