PX4/PX4-Autopilot #9185: Quad + Tail-Sitter crashed in FW position control

jingego, 2018-03-29:
Airframe: Quadrotor + TailsitterVTOL Quad Tailsitter (13004)
Hardware: PX4FMU_V2
Software Version: 0a58bd30
branch: master

[Flight log](https://logs.px4.io/plot_app?log=e6c1ad3b-140a-496c-98fa-12c5a0623aa7)
[Video](https://youtu.be/DBmkKT6kcoc)

Our vehicle was flying in FW mode around 6 mins. at the last turning, the pilot was accidentally adding some pitch down command when he gave tuning commands. The log does not record the moment that vehicle crashed. But the tlog shows that at least the autopilot was running. From the video, it can be seen that the pilot was already pulled up the vehicle (50s). but suddenly the aircraft pitch down again. we have checked the tlog,  the pilot pulled down the pitch stick and pulled up the throttle stick for the last seconds.  we are not able to figure out the reason that vehicle crashed at this moment.
 
when we reviewing the log, we found an unexpected airspeed sp that we do not quite understand:
[image]

The airspeed sp sudden dropped to the minimum airspeed that we set. I cut the video with the first sudden drop and the second sudden drop to crash in [Video segment](https://youtu.be/DLqRlTIKYKA). it can also be observed in our previous flight [(log)](https://logs.px4.io/plot_app?log=f65f5fa8-1407-45ca-926c-ee6c1437a76a).

[image]

The minimum airspeed sp looks related to the tecs.mode=1, which is [TECS_MODE_UNDERSPEED](https://github.com/PX4/Firmware/blob/d40d165b4b056f87528954ebcf85bf9a2992e7d9/msg/tecs_status.msg#L2). 

[image]

In my understanding of this mode, it should increase the airspeed sp instead of decreasing it [code].(https://github.com/PX4/ecl/blob/39b69af9bf8340790011c7c2723d4ed725ea0a3b/tecs/tecs.cpp#L322-L324)
could someone kindly help this?

itayguy, 2018-03-29:
Hi , like your data analyze software , can you share it ?

RomanBapst, 2018-03-29:
@jingego Thanks for sharing this! I'll have a look at your log to see if I can found a cause for the crash.

jingego, 2018-03-29:
@itayguy that is flightplot provide by PX4. you can download it [here](https://github.com/PX4/FlightPlot/releases)

RomanBapst, 2018-03-29:
@jingego Was this the first fw flight you've done?

jingego, 2018-03-29:
@RomanBapst for this vehicle, yes. For this airframe, we have tested a lot. This time we changed the power system.

RomanBapst, 2018-03-29:
@jingego Have you experienced before that the log stops mid air?

bkueng, 2018-03-29:
@jingego can you provide the tlog as well please?

jingego, 2018-03-29:
@RomanBapst experienced once a long time ago. I guess it stopped because crashed? (both are crashed). 
@bkueng sure! I uploaded it to Dropbox [Link](https://www.dropbox.com/s/qru5v7wsku668x2/2018-03-28%2013-11-09.tlog?dl=0).

jingego, 2018-03-30:
@RomanBapst is it possible the thrust decrease due to integrator wind-up problem? like issue #8508. The crashed vehicle use the firmware in 2017 Nov. And another vehicle [LOG](https://logs.px4.io/plot_app?log=3cef3f9b-8d14-4fa7-9db5-c3eb89b80f3d) do not have the thrust setpoints problem, which firmware is much newer, uploaded 2 weeks ago.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-09:
@jingego @RomanBapst is this actually resolved or is there something to follow up with?