PX4/PX4-Autopilot #9233: PX4 Ping Message Protocol

weiminshen99, 2018-04-03:
Inspired by #6721, has someone implemented the complete PING protocol among PX4s and QGC? The current code does only a part of the job: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1718-L1728

According to the spec in http://mavlink.org/messages/common#PING, a ping message should be either requesting or responding to a ping. This allows to measure the system latencies, including serial port, radio modem and UDP connections. It is particularly useful when multiple UAVs and PX4s are flying in a swarm formation. Given the fact that more and more people are interested in multiple-UAVs, could someone please take the lead and implement the complete protocol? Thank you very much in advance!

dagar, 2018-04-04:
Don't this just need the time and sequence filled in before broadcast?

mhkabir, 2018-04-04:
Yep. And a way to expose ping stats via `mavlink status`. 

I already have a working branch for this. Let me share it via a PR after some cleanup.

weiminshen99, 2018-04-06:
Hi, @mhkabir,
Thank you and that is great! I see you already made some enhancement milestones.
I do have a question about this protocol. According to the spec, which says that the target_system is "0: request ping from all receiving systems, if greater than 0: message is a ping response and number is the system id of the requesting system". If this is the only thing it does, then how do people ping from one particular UAV to another particular UAV?

mhkabir, 2018-04-06:
You can send a PING message with 0 and then check the returned PING for the sysid of the UAV you are interested in.

weiminshen99, 2018-04-06:
I see. Thanks!

mhkabir, 2018-04-07:
WIP branch is here : https://github.com/PX4/Firmware/tree/pr-ping-protocol. QGC does not seem to respond to pings, so I need help on that side to add support for returning pings. 

@DonLakeFlyer How hard would it be to add a PING response functionality in QGC?

dagar, 2018-05-27:
Implemented.

closed 2018-05-27 by dagar