PX4/PX4-Autopilot #9237: make is failing for Firmware cloned from version v1.6.5

Ankur1014, 2018-04-03:
Hi,
I have been using the PX4 Firmware version v1.6.5 for quite a while now. I cloned the Firmware by using the below command
`git clone --branch 'v1.6.5' https://github.com/PX4/Firmware.git`

Till 3 days ago, I was able to build the firmware and execute the  `make px4fmu-v3_default` command without any issue.

Today , the same command started failing showing the below log.

> CMake Error at CMakeLists.txt:447 (add_subdirectory):
  add_subdirectory given source "src/lib/DriverFramework/framework" which is
  not an existing directory.
 CMake Error at cmake/nuttx/px4_impl_nuttx.cmake:208 (get_target_property):
  get_target_property() called with non-existent target
  "lib__DriverFramework__framework".
Call Stack (most recent call first):
  src/firmware/nuttx/CMakeLists.txt:4 (px4_nuttx_generate_builtin_commands)
 CMake Error at cmake/nuttx/px4_impl_nuttx.cmake:208 (get_target_property):
  get_target_property() called with non-existent target
  "lib__DriverFramework__framework".
Call Stack (most recent call first):
  src/firmware/nuttx/CMakeLists.txt:4 (px4_nuttx_generate_builtin_commands)
 CMake Error at cmake/nuttx/px4_impl_nuttx.cmake:208 (get_target_property):
  get_target_property() called with non-existent target
  "lib__DriverFramework__framework".
Call Stack (most recent call first):
  src/firmware/nuttx/CMakeLists.txt:4 (px4_nuttx_generate_builtin_commands)

 I cloned the firmware from the release v1.6.5 again using the same git command again. THis time while running the make command I saw an error (when git was downloading the submodules).

>  Cloning into 'src/lib/DriverFramework'...
fatal: reference is not a tree: f80732f1b157c18f4b520b981f705b87f87a7ac8
Unable to checkout 'f80732f1b157c18f4b520b981f705b87f87a7ac8' in submodule path 'src/lib/DriverFramework'

Can anyone help ? I do not want to move the 1.7.4 right away and I want to continue in v1.6.5 for some more time.

dagar, 2018-04-03:
Could you try again? The DriverFramework submodule was (incorrectly) pointed at an old branch that was deleted. I've restored that branch and marked it protected.

closed 2018-04-04 by dagar

473867143, 2018-12-11:
Do you have a solution about this question ?
Today I meet the same question.i don't know how to do next?
can you give me some suggestion?
thank you a lot!

473867143, 2018-12-11:
-- CMAKE_MODULE_PATH: /home/px4/src-1.6.5/Firmware/cmake
-- Nuttx build for px4fmu-v2 on m4 hardware, using nsh with ROMFS on px4fmu_common
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5
-- CONFIG: nuttx-px4fmu-v2-default
-- The ASM compiler identification is GNU
-- Found assembler: /home/px4/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
CMake Error at cmake/nuttx/px4_impl_nuttx.cmake:266 (add_subdirectory):
  add_subdirectory given source "/home/px4/src-1.6.5/Firmware/nuttx-patches"
  which is not an existing directory.
Call Stack (most recent call first):
  cmake/nuttx/px4_impl_nuttx.cmake:659 (px4_nuttx_add_export)
  CMakeLists.txt:346 (px4_os_prebuild_targets)

-- Adding ROMFS on px4fmu-v2
-- Nuttx build for px4fmu-v2 on m3 hardware, using nsh 
CMake Error at cmake/nuttx/px4_impl_nuttx.cmake:266 (add_subdirectory):
  add_subdirectory given source "/home/px4/src-1.6.5/Firmware/nuttx-patches"
  which is not an existing directory.
Call Stack (most recent call first):
  cmake/nuttx/px4_impl_nuttx.cmake:659 (px4_nuttx_add_export)
  src/modules/px4iofirmware/CMakeLists.txt:50 (px4_os_prebuild_targets)

-- Configuring incomplete, errors occurred!
See also "/home/px4/src-1.6.5/Firmware/build_px4fmu-v2_default/CMakeFiles/CMakeOutput.log".
/bin/sh: line 0: cd: /home/px4/src-1.6.5/Firmware/build_px4fmu-v2_default: No such file or directory
Makefile:164: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

dagar, 2018-12-11:
@473867143 try `make distclean` then `make px4fmu-v2_default`

473867143, 2018-12-20:
@dagar i have solved,thanks a lot