PX4/PX4-Autopilot #9240: Frsky telemetry Bug?

kuiskas, 2018-04-04:
Dear community.
I have an Taranis and a X8R receiver connected to my PX2.1 through one of the craftcand theory cables.
http://www.craftandtheoryllc.com/store/ct-telemetry-cable-for-frsky-radios/
I tried everything in order to get telemetry over SPORT but not working.
Enable in Telemtry2 SYS_COMPANION as FrSky, task is started and working (frsky_telemetry status SPORT) but I get not telemetry on the Taranis, I have followed the instruction here:
https://docs.px4.io/en/peripherals/frsky_telemetry.html
 
My question: Does it work from everybody? What could I posible did wrong?

Thanks in advance for the help.

bkueng, 2018-04-05:
Generally works fine for me. Did you do the sensor discovery as described here: https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry?
If `frsky_telemetry status` shows SPORT, the problem is likely on the taranis side.

kuiskas, 2018-04-06:
thanks bkueng,
I managed to get it  working on /dev/ttyS6 (GPS2).
But I neither success on ttyS2 (telemetry 2) nor in ttyS7 (Serial 5, CONS). I do not know how to enable this port. 
Do you get it over ttyS2 or ttyS7?

kuiskas, 2018-04-09:
Ok I will search it, and when I find it I will post the solution. For the moment I can live using ttyS6 (GPS2)
Thanks and closing.

closed 2018-04-09 by kuiskas

hamishwillee, 2018-04-11:
@bkueng Our docs on https://docs.px4.io/en/peripherals/frsky_telemetry.html seem a bit "thin" - for a start they only cover pixracer and they do it in only the briefest of terms:

> For receivers with D-port (D4R-II and similar): Connect the Pixracer FrSky TX line to the D4R-II RX line. Also connect GND.

Suggestions/thoughts:
1. Add images from or link to https://docs.px4.io/en/assembly/quick_start_pixracer.html#radioremote-control
2. Do we need to add that link you pointed to: https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry ?
3. What about for Pixhawk and other boards - we don't say much about this in https://docs.px4.io/en/assembly/quick_start_pixhawk.html#radio-control

What I like to see here is idiot proof instructions:
- Explanation of what ports you can use for connecting
- Explanation of what you might have to enable/change in configuration
- Images showing connections

Thoughts?

bkueng, 2018-04-11:
> Add images from or link to https://docs.px4.io/en/assembly/quick_start_pixracer.html#radioremote-control

yes.

> Do we need to add that link you pointed to: https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry ?

That would probably help, yes. It's just the first link I found, and there are video tutorials as well, for instance https://www.youtube.com/watch?v=x14DyvOU0Vc.

> What about for Pixhawk and other boards - we don't say much about this in https://docs.px4.io/en/assembly/quick_start_pixhawk.html#radio-control

You can set `SYS_COMPANION` to `10`, and connect to the TELEM2 UART. Alternatively on pixhawk pro, you can connect it to TELEM4 (no config needed). If you add it to the startup script yourself, you can run it on any UART.