PX4/PX4-Autopilot #9242: px4fmu-v2_default won't build with latest master

DonLakeFlyer, 2018-04-04:
* make distclean
* make px4fmu-v2_default

```
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: address 0x8102274 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: nuttx_px4fmu-v2_default.elf section `.ARM.exidx' will not fit in region `flash'
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: address 0x8102274 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: address 0x8102274 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 12341 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

jlecoeur, 2018-04-05:
Hi, can you try with a newer compiler? 
The doc recommends gcc-arm-none-eabi-7-2017-q4-major: https://dev.px4.io/en/setup/dev_env.html

taileron, 2018-04-05:
An easy way to for a quick change of the compiler version is to find the GCC folder substructure on the machine e.g. /usr/local/Cellar/gcc-arm-none-eabi/ or add a separate (visible) own system path. There have to be 4 folders containing the whole GCC (arm-none-aebi;bin;lib and share) they can be replaced by the desired GCC (7.2.1) version

taileron, 2018-04-05:
... these folders are invisible, they have to be visible first then create an alias from folder "20160928" and copy it e.g. to the desktop afterwards the the file system can be set to invisible again.
And the desired compiler folders can be copied into the alias folder that automatically remains visible (I often change between different GCC versions this way)

DonLakeFlyer, 2018-04-05:
If I redo toolchain installation will it update compiler?

LorenzMeier, 2018-04-05:
Yes.

taileron, 2018-04-05:
I think on unix platforms like e.g. mac os the toolchain installer only contains the mentioned 4 folders that have to be copied to the appropriate location.
Only the whole homebrew procedure overwrites GCC in the right way.

DonLakeFlyer, 2018-04-05:
So I redid the toolchain on osx. It told me I needed to ```brew upgrade``` so I did that. Then it told me a bunch of stuff was already installed. So I did ```reinstall``` just to be safe.

```

taileron, 2018-04-05:
so the only way is overwrite the 4 folders located in usr/local/Cellar/gcc-xx/20160928 with the new GCC folders that the toolchain 7.2.1 offers.
with a "hidden files widget" hidden paths can be shown

DonLakeFlyer, 2018-04-05:
Great! Now all of this toolchain updating has trashes my ability to build gazebo sitl:
```
/usr/local/include/ignition/msgs0/ignition/msgs/color.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/color.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
```

```make distclean``` does not help

DonLakeFlyer, 2018-04-05:
jmavsim is trashed as well:
```

DonLakeFlyer, 2018-04-06:
I blew away my homebrew complelely and reinstall toolchain and gazebo is working.

So for this specific issue, it seems like a bug that toolchain installation is not updatign compilers?

julianoes, 2018-09-10:
@DonLakeFlyer I'm assuming the v2 build has been fixed again?

For the Java issue, let's track it in #10443.

closed 2018-09-10 by julianoes