PX4/PX4-Autopilot #9260: Crash: Pixhawk 2.1 (Cube) with v1.7.4 Beta firmware

r0gelion, 2018-04-06:
Flying an F450 with Pixhawk 2.1 (Cube) and Here GPS, loaded v.1.7.4 Beta (e63f9d9).

The pilot reported the vehicle ended a mission and it was hovering in position hold and suddenly it flew to a side in a straight line descending and it crashed on a roof.

Here is the log:
https://review.px4.io/plot_app?log=34c46166-8fee-4cb8-b7de-38517de4122b

Thank you!

RomanBapst, 2018-04-09:
@r0gelion The log ends when the pilot was maneuvering the drone in position control mode. Until then I cannot see any strange behavior, if you actually use the 3D view you can see that the vehicle followed the pilot's commands very well.
Is it possible for the pilot to have a look at the 3D view and maybe estimate the time between when the log stops and when the vehicle started to drift away?

RomanBapst, 2018-04-09:
@r0gelion Do you happen to have a mavlink log of this flight?

r0gelion, 2018-04-10:
@RomanBapst this is the log:

[2018-04-05 12-35-45.zip](https://github.com/PX4/Firmware/files/1896718/2018-04-05.12-35-45.zip)

dagar, 2018-05-02:
Could someone go over the timeline? How much longer do we think the vehicle was in the air after the log stopped?

We only have about 2 seconds of RADIO_STATUS data from the vehicle after the last regular mavlink message.

[image]

dagar, 2018-05-02:
What was the state of the vehicle after the crash?