PX4/PX4-Autopilot #9263: Allow initialisation without external vision (rotate external vision)

Changliu52, 2018-04-07:
Hi @mhkabir @dagar @priseborough ,

One more question here.

I now have the ekf setup as:
EKF2_AID_MASK enables `use optical flow`, `vision position fusion` and `rotate external vision`;
EKF_HGT_MODE set to `Vision`.

I wonder whether it would make more sense to allow local_position to be initialized without external vision at startup? It should be sufficient to use optical flow and baro to initialize the local_position, and when external vision pose is initiated during flight, it resets the offsets and rotate external vision to start fusion the vision pose.

I think this is useful, because most visual SLAM doesn't initialize without moving, but the PX4 doesn't initialize the local position without external vision. So manually moving the drone is always required at startup. This law makes sense if `vision yaw fusion` is enabled, but from my understand this is no longer the case if `rotate external vision` is enabled instead.

Please let me know your thoughts.

Kind Regards,
Chang

yc718764666, 2018-06-09:
hello, could you solve this problem?  I try ti use the  Gazebo‘s Force and Torque tool to move the UAV for which can be initialized, but the result doesn't work well ,the initialization is aways defeated.

Changliu52, 2018-06-11:
Currently, I just sent out a bunch of fake vision_pose to pass the initialisation check for px4.
This only work when `rotate external vision` and `vsion yaw fusion` are disabled, and I rotate external vision in ros before passing into the vision_pose estimation.

mhkabir, 2018-06-20:
Let's fix this now. I'm having a look at what needs to be done.

Changliu52, 2018-07-26:
Sorry for the delay. I was in the middle of a demo, so didn't have time to test.
Thank you so much for the update @mhkabir . It seems the link is not available anymore?

Thanks

mhkabir, 2018-07-26:
It has been merged to ecl master.

closed 2018-07-26 by mhkabir

Changliu52, 2018-07-26:
Nice. Thanks. Kabir. I will do some further testing.

Changliu52, 2018-07-27:
Hi @mhkabir ,
Please see the flight log here with the firmware at 57dfcee8423babe009110b313e131d5672529938
Handholding test. Link here: [https://review.px4.io/plot_app?log=7d6c9ab6-e5fc-4e21-86c8-991b27d3518a](url)

EKF2_AID_MASK: vision position fusion + rotate external vision
EKF2_HGT_MODE: vision
EKF2_EV_DELAY: 0.0ms
EKF2_ASP_DELAY: 250ms

1) The local pose took several attempts at the beginning to follow vision correctly, and at the same time it complained about "EKF ev hgt timeout" and switch between vision and baro. It doesn't make sense because vision x, vision y are ok, only vision z has time out?
2) when switching baro, the spike problem in local position z came back. I think there are still some bugs for resetting the EKF at switching.
3) after several attempts, it did start to follow vision with an offset transform (which is good), however the local pose tends to wonder around when the motion evolve significant yawing. Was this caused by the internal vision offset estimation?

Please let me know your thoughts.

Best Regards,
Chang

Changliu52, 2018-07-27:
Please see the same question in a new issue thread: https://github.com/PX4/Firmware/issues/10061