PX4/PX4-Autopilot #9266: Discussion: Look into improved curve fits for ms5611 Baro temp calibration

Antiheavy, 2018-04-08:
The 5th order curve fit for the ms5611 Baro sensor sometimes does a poor job at cold temperatures.

We've been performing the offline temp calibration process on a few handfuls of autopilots and they all look fairly similar to this plot:

[image]

Our concern is the diverging curve fit at the cold temps.  

Also, the ms5611 has a recommended temp compensation scheme which may be implemented somehow either inside the ms5611 or inside the PX4 drivers (not clear to me).  There may be some confusion or conflicts with the off-board calibration and the chip/driver level compensation.
[image]

Improvements might involve:
- better handling of temperatures outside the calibrated limits.  what should be done here?  better/safer curve fits or extrapolations? warnings to the user?  prevent arming until within calibrated limits? ( I'm not sure preventing operation outside calibration limits makes sense as it may force users to have expensive temp chambers to do wide temp calibrations.)
- better coordination between the ms5611 recommended temp compensation and the actual raw data calibration curves generated by PX4 (onboard and/or offline).

Thoughts?
@dagar @priseborough @M-Skelton

M-Skelton, 2018-04-08:
Here are a few more examples for reference:

SN0077:
[image]

SN0075:
[image]

SN0072:
[image]

Another Senthawk:
[image]

A pixracer:
[image]

These calibrations were all done using the exact same method to perform the off-board calibration. 
1. set calibration parameters
2. cold soak for 1 hr. 
3. remove from freezer and connect usb cable
4. place into cooler 
5. seal cooler 
6. connect usb cable to the computer 
7. let the calibration run for 30 min 
8. run the of board python scripts 

I think it's interesting that we see similar curves for 0075, 0072, and the developer Senthawk, but very different curves for the pixracer and 0077. 

I think there's also something to address with:
https://github.com/PX4/Firmware/pull/9054

I posted in the forums about the same issue a while ago for reference: 
http://discuss.px4.io/t/mpu9250-internal-temperature-reading-seems-to-be-incorrect/5542/2

I think before we can roll this calibration into a production standard we need to ensure the sensors are reading correctly in the first place. As well as getting more consistent results. 

As for how the firmware/autopilot should interpret past the calibrated range, I think reverting back to using the raw sensor reading might be a good idea for now. The curves that I posted hint at another inflection at the lower range that is shown. So I don't a linear interpolation out past the calibrated range would be of much use or a good idea. 

If we could come up with a clever way to try and merge the raw and calibration data that would be neat, but as for the mechanics of how to implement that in a useful way, I don't have much of an instinct for at this moment.

priseborough, 2018-04-08:
The manufacturers recommended compensation scheme is supposedly being implemented inside the Baro driver. This needs to be checked. If we are doing our own serial number specific sensor calibration downstream of the driver then we may be better off not doing the model number specific calibration provided by the manufacturer.

dagar, 2018-04-08:
Pull request for testing https://github.com/PX4/Firmware/pull/9267.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
@dagar re-open if this is still going to happen.