PX4/PX4-Autopilot #9283: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice

medhijk, 2018-04-10:
When I try to run the jMAVSim, I run into the following error. I am using Ubuntu 16.04. Please help!

`medhijk@asimov:~/Firmware$ make posix jmavsim
[2/2] cd /home/medhijk/Firmware/build/...hijk/Firmware/build/posix_sitl_default
args: /home/medhijk/Firmware/build/posix_sitl_default/px4 posix-configs/SITL/init/ekf2 none jmavsim none /home/medhijk/Firmware /home/medhijk/Firmware/build/posix_sitl_default
SITL ARGS
sitl_bin: /home/medhijk/Firmware/build/posix_sitl_default/px4
rcS_dir: posix-configs/SITL/init/ekf2
debugger: none
program: jmavsim
model: none
src_path: /home/medhijk/Firmware
build_path: /home/medhijk/Firmware/build/posix_sitl_default
empty model, setting iris as default
SITL COMMAND: /home/medhijk/Firmware/build/posix_sitl_default/px4 /home/medhijk/Firmware /home/medhijk/Firmware/posix-configs/SITL/init/ekf2/iris
data path: /home/medhijk/Firmware
commands file: /home/medhijk/Firmware/posix-configs/SITL/init/ekf2/iris
250 WARNING: setRealtimeSched failed (not run as root?)

| ___ \ \ \ / / / |
| |/ / \ V / / /| |
| __/ / \ / /| |
| | / /^\ \ ___ |
_| / / |_/

px4 starting.

INFO [dataman] Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 11405132 bytes
INFO [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..
Buildfile: /home/medhijk/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 1 second
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x4bab7e41, isOwner false, <37eca4c6, 580b55d5>[count 0, qsz 0, owner ]]]
at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
... 2 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
at java.lang.reflect.Method.invoke(Method.java:498)
at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped
at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2047)
at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1917)
at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1866)
at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1835)
at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:79)
at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:229)
at java.security.AccessController.doPrivileged(Native Method)
at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:215)
at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2272)
at com.jogamp.opengl.GLProfile.get(GLProfile.java:1020)
at com.jogamp.opengl.GLProfile.get(GLProfile.java:1049)
at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:802)
at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:132)
at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:837)
at javax.media.j3d.VirtualUniverse.(VirtualUniverse.java:274)
at me.drton.jmavsim.Visualizer3D.(Visualizer3D.java:154)
at me.drton.jmavsim.Simulator.(Simulator.java:157)
at me.drton.jmavsim.Simulator.main(Simulator.java:678)
... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7fec7400d080, refCount 1, unCloseable false]
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x218d7dcf, isOwner false, <6b69ecc8, 5534c0e8>[count 0, qsz 0, owner ]]]
at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
... 2 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
at java.lang.reflect.Method.invoke(Method.java:498)
at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped
at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2047)
at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1917)
at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1866)
at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1835)
at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:79)
at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:229)
at java.security.AccessController.doPrivileged(Native Method)
at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:215)
at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2272)
at com.jogamp.opengl.GLProfile.get(GLProfile.java:1020)
at com.jogamp.opengl.GLProfile.get(GLProfile.java:1049)
at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:802)
at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:132)
at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:837)
at javax.media.j3d.VirtualUniverse.(VirtualUniverse.java:274)
at me.drton.jmavsim.Visualizer3D.(Visualizer3D.java:154)
at me.drton.jmavsim.Simulator.(Simulator.java:157)
at me.drton.jmavsim.Simulator.main(Simulator.java:678)
... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 1/1, reusable (open, marked uncloseable): 0, pending (open in creation order): 1)
X11Util: Open X11 Display Connections: 1
X11Util: Open[0]: NamedX11Display[:0, 0x7fae2400b210, refCount 1, unCloseable false]`

hamishwillee, 2018-04-10:
I don't know the cause, but have you set up the developer environment as covered in https://dev.px4.io/en/setup/dev_env_linux.html ?

medhijk, 2018-04-10:
@hamishwilee : Yes, I am pretty sure I followed those instructions to set up the development environment in linux. I may be wrong but I think it has something to do with display and I am not sure where to fix it

hamishwillee, 2018-04-10:
I just ran the instructions from scratch on a clean Ubuntu 16.04 (running in VirtualBox VM on windows) using ubuntu_sim_common_deps.sh (https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html). All just worked.

I don't know why this would fail for you sorry!

malikaoudjif, 2018-04-15:
i have the same problem, any solution

hamishwillee, 2018-04-15:
@malikaoudjif Again, id you run the script in ubuntu 16.04 "clean"? Perhaps post the whole build log to a gist and link here?

Cody-Chen, 2018-04-21:
please read my comment: https://github.com/DrTon/jMAVSim/issues/42
I have solve it. Maybe file a PR for you.

aogrcs, 2018-10-24:
@hamishwillee Hi, what's your mean of clean Ubuntu 16.04, I run it in VMwrae on windows. I open it, update and upgrade, the using ubuntu_sim_common_deps.sh, but there is the same error. Somebody said that it might be caused by the version of opengl, which should be 3.1 or higher, but it was 3.0 now? Thanks

hamishwillee, 2018-10-25:
@aogrcs I have no idea. BUt it looks like @Cody-Chen might: https://github.com/PX4/Firmware/issues/9283#issuecomment-383273485

aogrcs, 2018-10-30:
Hi @hamishwillee Thanks
I thought the error might be caused by using make in the directory Firmware not the /src/Firmware.
I did make in the directory Firmware which I cloned, there was the error; but I used the shell and did make,  it worked.

hamishwillee, 2018-10-30:
@aogrcs Definitely not the location of firmware. I would have guessed something to do with opengl. Did you have to do anything other than standard build instructions and  `make posix jmavsim` to get this to work?

aogrcs, 2018-10-30:
@hamishwillee I will test it for complete new image, If I got result, I would post it. Thanks

hamishwillee, 2018-10-31:
THanks. I tried this some time ago and it didn't work - would be great if this had somehow fixed itself.

aogrcs, 2018-10-31:
Hi @hamishwillee I did the test. I used a clean Ubuntu 16.04 64, after installing it I did not use update or upgrade. Then I run ubuntu_sim_common_deps.sh, got the Firmware clone. In the directory, "make posix_sitl_default jmavsim" worked except to install pyyaml manually. I also got Firmware using git and it also worked, so the location of the Firmware did not make the problem. I think I made mistake before. Thanks

metinburak, 2018-10-31:
@aogrcs  when you clone FW and make posix for sitl jmavsim, it works but after you upgrade 16.04, for me it doesn't work again with the same error. I removed my graphics card from pciex slot and used onboard graphics adapter and it worked again. It has got something to do between a new package which is updated for the display and the graphics card adapter. It has got nothing to do with the PX4 firmware I think.

aogrcs, 2018-11-01:
@metinburak Hi, could you give more information about your error?

hamishwillee, 2019-01-30:
I believe this is indicative of a driver error. It occurs in VMs, and often enough in real hardware that we should have as a troubleshooting guide?

@bkueng What can users do if they get this error? Just update to latest drivers for all their software and pray? :-)

closed 2019-02-13 by stale[bot]

PimoGithub, 2020-04-30:
I have the same issue on my thinkpad where I just installed Ubuntu 20.04
I can start the simulator by setting "MESA_LOADER_DRIVER_OVERRIDE=i965"
So my command line is:
> MESA_LOADER_DRIVER_OVERRIDE=i965 ./Tools/jmavsim_run.sh -q -d /dev/ttyACM0 -b 921600 -r 250 -o

I found this fix here: https://www.mathworks.com/matlabcentral/answers/342906-could-not-initialize-shared-resources-for-x11graphicsdevice

bresch, 2020-05-20:
Had the same issue on my Thinkpad, and the solution of @PimoGithub worked for me as well:
`MESA_LOADER_DRIVER_OVERRIDE=i965 make px4_sitl jmavsim`

MaEtUgR, 2020-05-20:
Probably latest Linux distros switched the Intel driver architecture to Iris https://linuxreviews.org/Intel_Iris
And the old java graphics library jmavsim uses is not properly supported anymore. Just a guess.