PX4/PX4-Autopilot #9288: A litttle upload problem.

KJNKJNKJN, 2018-04-11:
[image]

When upload the program,  shell will ends with the above picture. Although it succeed,but i really want to know how to end with this output:
Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.

[100%] Built target upload

Thank you!

mcsauder, 2018-05-19:
Hi @KJNKJNKJN , Does this stand out as an issue to you from your screen capture above?  I would seem to think that this could be part of your problem.  Perhaps not?

[image]

mcsauder, 2018-05-19:
Hi @KJNKJNKJN , Can you post your repo branch, version, etc. ?

julianoes, 2018-06-26:
What exactly is the issue. I'm trying to understand.

KJNKJNKJN, 2018-06-29:
@julianoes  @mcsauder  the PX4 Firmware version is 1.7. When download the programme, it does not end with output:
Erase : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.

[100%] Built target upload

.Althougth the upload succeed. thank you.

mcsauder, 2018-06-29:
Hi @KJNKJNKJN ,
 
Please try this, delete your clone of the PX4/Firmware repository, re-clone it, complete a 
```
git submodule update --init --recursive
```

then, depending on your particular hardware configuration, complete something that matches your hardware version such as:

```
make px4fmu-v4_default upload
```

and if your hardware is v2, substitute v2 instead of v4, etc., ...

Next, report back to us.  We are very interested in helping you, please feel free to DM us on slack to get through this, we have all been in your position previously.  Good luck!

-Mark

KJNKJNKJN, 2018-06-29:
@mcsauder thank you very much.  I try it.

mcsauder, 2018-07-13:
@KJNKJNKJN , 

Any luck?  If you have had success shall we move to close the issue?

-Mark

closed 2018-07-15 by KJNKJNKJN