PX4/PX4-Autopilot #9291: crazyflie 2.0 cannot get local position from mavros

vuanhduy, 2018-04-11:
**Question**

Hi all,
I am following the instruction from https://docs.px4.io/en/flight_controller/crazyflie2.html and successfully installed px4 firmware onto my Crazyflie 2.0.

I can also run `cfbrigde` (both C and python versions) application and `mavros` to communicate with the drone. I have checked the `/diagnostics` topic  and can retrieve data from `/mavros/imu/data`, /`mavros/battery`, `/mavros/altitude` and `/mavlink/from` topics via the `rostopic echo` command.

However, when I publish data to` /mavros/vision_pose/pose` topic, I can only obtain the local position via Mavnlink Inspection feature of the QGroundControl (i.e., the `LOCAL_POSITION_NED` section correctly shows the information) but I cannot get it by `rostopic echo /mavros/local_position/pose` command. Why?

@dennisss, @barzanisar: I see that both of you also work with the crazyflie so could you give any suggestion?

barzanisar, 2018-04-12:
Are you able to echo the topics as well as see them in QGC at the same time?

barzanisar, 2018-04-12:
Did you connect the crazyflie with a USB cable to your PC as well? or just via crazyradio?

vuanhduy, 2018-04-12:
@barzanisar: thank you for your reply. I have updated the mavlink option allowing to stream the LOCAL_POSITION_NED. Just my curiosity, are you also using crazyflie 2.0 for an indoor project?

closed 2018-04-12 by vuanhduy

thomasgubler, 2018-04-12:
➤ Barza Nisar commented:

yes I am.
Are you able to echo the topic now?

thomasgubler, 2018-04-12:
➤ Barza Nisar commented:

Also, were you able to echo the topic as well as use QGC at the same time? Or echoing stops when you switch on QGC?

vuanhduy, 2018-04-12:
Yes, I am able to echo the topic now, and I can do both (echoing and QGC) at the same time. Are you using offboard mode to fly the drone indoor?

barzanisar, 2018-04-12:
No, I am using other modes for now. I might use offboard later on. Why?

I was not able to echo the topic as well as use QGC at the same time. Which cfbridge are you using? C based or python script? and do you launch mavros listening to the same udp port as QGC? i.e. 14550
by running `roslaunch mavros px4.launch fcu_url:="udp://:14550@127.0.0.1:14555"` ?

vuanhduy, 2018-04-12:
Hi @barzanisar,
I am using the python version and created a launch file for it which is just a copy of px4.launch setting `fcu_url` to `udp://:14550@127.0.0.1:14556` and `gsc_url` to `udp://@192.168.0.102:14557`. (I am running the QGroundControl app on 192.168.0.102 machine and let it listen the port 14557.)

However, I'm getting some difficulties in running offboard mode. The offboard mode requires publishing the target NED position every 20 milliseconds but the drone sometimes disconnects with the radio dongle for more than 20 milliseconds which makes the drone escape the offboard mode and be out of control.