PX4/PX4-Autopilot #9295: Toile Bowl

albert21martins, 2018-04-12:
**Bug Report** 
Hi,

We have a 5 kg coaxial octocopter running with Pixhawk 2.1, Here Gps and version 1.7.3 stable.
We are experiencing some toilet bowl behavior in Position mode. Last week was running everything excellent, than we had a small crash, related to takeoff ramp, and since then, the copter is experiencing  toilet bowls and fly aways.
We are getting no error, but after switching back from position to alti the copter is drifting and flying away. We tried to minimize the cable and the components, re calibrated a lot of times and noting.
here one of the logs. May be is it helpful
https://logs.px4.io/plot_app?log=20d49b5c-de94-4186-95b4-dd74c27b3bc5

Thanks in advance,

Albert

bresch, 2018-04-19:
Hi @albert21martins 
Usually, the toilet bowling behavior is due to bad mag calibration/magnetic field interferences (misalignment of the measured heading with the true heading) . You also mentioned in the log file that you don't have an external compass. I've heard that flying with the internal compasses is really challenging that's why I would recommend you to add an external compass module.
The vibrations on your platform are also quite high, this can also cause a position/velocity estimation problem. It could be helpful to check if the props are properly balanced and maybe improve the damping of the autopilot.

albert21martins, 2018-04-22:
Hi Bresch, 
thanks for the answer. At the end it seems that the vibrations on the system were to high, already solved, was because the props were to tight (folding props)... Thanks again for the typ.

closing

closed 2018-04-22 by albert21martins