PX4/PX4-Autopilot #9297: A New External IMU Containing ADI Sensors Has Been Released to Support PX4 Firmware

zarathustr, 2018-04-13:
Hi Guys!

Recently we released one new external IMU with ADI analog inertial sensors in China. The sensor readouts are sampled at 10KHz via an onboard ADC. Later processing like calibration of temperature drift, biases, misalignments, factors has been conducted in-factory. The current price now is 1999 RMB (317.7707 USD) for basic version and 2999 RMB (476.7355 USD) for medium version. The advanced version will be released soon which contains onboard INS/GPS/Baro/Vision fusion supporting more robust navigation for flights.

We are happy that the PX4 team welcomes the new product and we will upload the open-source driver of the IMU soon. The code reads the IMU at the frequency of 1000Hz via the SPI connection with software CRC support according to relatively long-range transmission. The IMU has been tested to be much more effective than low-grade ones like MPU6000, ICM-20689 and etc. on the real-world electronic or gasoline quadrotors, where the mechanical jitters are quite distinct. It has also been tested with a 10-meter-wide fixed-wing plane for long-endurance flight.

The English version website is now under construction and anyone who interested can contact me directly or turn to Jinger Zeng for further acquisition. Thanks!

LorenzMeier, 2018-04-15:
Looking forward to it!

zarathustr, 2018-04-22:
The flight experience has been improved. As is known, the Pixhawk, Pixhawk 3 Pro and Pixhawk 4 all have onboard external SPI connector. The IMU is applied to these platforms with sampling frequency of 1000Hz. 

The following is a typical sampling via the PX4 with 3 ADXRS620 gyros
[image]

With the onboard MPU6000 and external IMU, two test flights have been conducted. For the first one, the MPU6000 is used for computing attitude. The control performances are shown as follows:

[image]

[image]

[image]

[image]

**While for the external IMU, the performances are:**

[image]

[image]

[image]

[image]

Note that here during the two flights, the parameters are default and the quadrotor is operated at the same position so the comparison is fair. It is obvious that the attitude rates control using ADI IMU is better which leads to better controlling behaviour on the roll and pitch. The basic version containing 3 ADXRS620 gyros and two ADXL203 accelerometers will be priced at 1999 RMB (317.7707 USD) while for the medium version it would be 2999 RMB (476.7355 USD) which replaces the yaw gyro with ADXRS642. **All the sensors have been factory calibrated** including factors, misalignment, biases and temperature drift. We hope that this is going to improve the flight experience in the future!

PS: The test PX4 ulog files can be downloaded on: [website](https://github.com/zarathustr/ext_imu_test)

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-10:
Please re-open if something needs to be done here.