PX4/PX4-Autopilot #9300: Tailsitter mission problem

RomanBapst, 2018-04-13:
@dagar FYI
Seems like tailsitter mission and position control is broken on master (it does not track at all, roll setpoints changes sign frequently).
I found a working version not too long ago and will do a bisect.

dagar, 2018-04-13:
Let's get a simple tailsitter mission added to CI after this is fixed.

dagar, 2018-04-13:
@RomanBapst I won't have time to look into this today, but I have a hunch this could be the problem. 

https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L706-L718

If that's the case we really should centralize the tailsitter frame conversion directly in the attitude update method rather than having them spread out and done as needed.

dagar, 2018-04-14:
WIP PR to add tailsitters to CI mission testing. https://github.com/PX4/Firmware/pull/9307