PX4/PX4-Autopilot #9316: Gazebo HIL simulation broken

Mtbarros, 2018-04-17:
**Bug Report**

Hello,

I tried running the Gazebo HIL simulation with a Pixhawk, exactly as detailed in the [guide](https://dev.px4.io/en/simulation/hitl.html), with the 1.7.3 stable release, but got this error:

WARN  [mavlink] no ack from logger (is it running?)
INFO  [logger] Opened log file: /fs/microsd/log/sess011/log001.ulg
INFO  [logger] Start mavlink log
ERROR [logger] Ack timeout. Stopping mavlink log
WARN  [commander] Failsafe enabled: no global position

After this the drone wouldn't even fly, or just crash.
Here is a log of the "flight": https://logs.px4.io/plot_app?log=0770a288-1584-4a1d-ab32-3e797312fd2a
(I mistankenly captured too much time after the drop, which is at about 0:51s)

When trying the 1.7.4 beta release it doesn't work because of a change in cmake/configs/nuttx_px4fmu-v2_default.cmake [in this commit (22acb1da)](https://github.com/PX4/Firmware/commit/22acb1dab1483fcbf618a8836864750ff8596b8f). Is there any special reason for deactivating pwm_out_sim? Or can I enable it in my own config and compile a new version of the firmware?

Thanks for your time

Mtbarros, 2018-04-17:
Got a fast response to the beta release change:

https://github.com/PX4/Firmware/commit/22acb1dab1483fcbf618a8836864750ff8596b8f#r28613835

So I will just compile the 1.7.4 with the pwm_out_sim module enabled and report if it is working as expected.

julianoes, 2019-07-01:
This should work with 1.9, closing.

closed 2019-07-01 by julianoes