PX4/PX4-Autopilot #9318: Simulated RC via QGC is broken

RomanBapst, 2018-04-17:
Setting the parameter COM_RC_IN_MODE to 2 (Virtual RC by Joystick) does not work anymore.

dagar, 2018-04-17:
I think we should get rid of this.

dagar, 2018-04-17:
To clarify, this is how it currently works

COM_RC_IN_MODE
 * 0
   - commander manual input checks will run
   - manual input comes from the first of publisher of manual_control_setpoint (either RC input or mavlink MANUAL_CONTROL)
   - Mavlink MANUAL_CONTROL -> uORB manual_control_setpoint
 * 1
   - commander manual input checks will **NOT** run
   - manual input comes from the first of publisher of manual_control_setpoint (either RC input or mavlink MANUAL_CONTROL)
   - Mavlink MANUAL_CONTROL -> uORB manual_control_setpoint

 * 2
   - commander manual input checks will **NOT** run
   - Mavlink MANUAL_CONTROL -> uORB input_rc (fake RC pwm)-> manual_control_setpoint (regular sensors RC path)

dagar, 2018-04-17:
I propose deleting the Mavlink MANUAL_CONTROL -> uORB input_rc path (current COM_RC_IN_MODE 2) entirely, then cleanup COM_RC_IN_MODE so that it makes a bit more sense.

I also think it would help if we didn't automatically take manual input from any available source. `COM_RC_IN_MODE 0` would mean RC input only, etc.

Later I'd like to extend this so that it's possible to safely handoff between regular RC and Mavlink manual input, but that's another discussion.

dagar, 2018-04-17:
It's also worth being aware of RC_CHANNELS_OVERRIDE. At the moment it's possible for this message to interfere with actual RC (something we should fix).

https://github.com/PX4/Firmware/blob/64aab8a685e29a5f2d267d7574db3da49317a30b/src/modules/mavlink/mavlink_receiver.cpp#L1567-L1622

RomanBapst, 2018-04-17:
@dagar I did some debugging and and the problem seems to be that at the very beginning the mavlink receiver claims the first instance of manual_control_setpoint but soon stops to publish as it shouldn't. The sensors module then gets the second instance of the topic to which nobody subscribes and it seems manual control is lost.

RomanBapst, 2018-04-17:
@dagar The main reason why I liked to use COM_RC_IN_MODE 2 is because it allows to assign switches as you would do if you had a standard RC connection. Currently it's not possible to assign a transition switch if you use the joystick mode.

dagar, 2018-04-17:
If there's more followup let's discuss this on a call. Getting manual control without RC right is important for commercial adoption. This is an area where some proprietary flight controllers I've used are still much better.

1. Flying with a joystick or gamepad (real world, SITL, HITL)
  * Mavlink MANUAL_CONTROL -> uORB manual_control_setpoint
  * mode switches, transition commands, etc can be mapped to buttons on the controller and sent as mavlink commands (with acknowledgment)
  * @RomanBapst was transition the only thing you were missing here? I've added this as an option before, but apparently it never made it upstream.

2. Testing full RC in SITL or without RC equipment.
  * Mavlink RC_CHANNELS_OVERRIDE -> uORB input_rc
  * if the desire is to test as much of the RC/input pipeline as possible (without physically having RC) this is a much better fit

What are we missing that can't be satisfied by 1 & 2?

One minor thing that's come up before is the behaviour of a gamepad in a situation with lossy comms. I believe including the state of all buttons/switches in every MANUAL_CONTROL is better, or at least more closely resembles how RC behaves than sending discrete mode change commands that queue.

I really think we need to be consolidating things like this into a single good solution wherever possible. It minimizes the test area and ensures we're using the system like real world operators.

tubeme, 2018-04-18:
I recently made tests with 3.3 and added some issues:

https://github.com/mavlink/qgroundcontrol/issues/6265

https://github.com/mavlink/qgroundcontrol/issues/6263

Don closed these two because they are related to PX4 and not to QGC

And there are three issues open in the ggc:

https://github.com/mavlink/qgroundcontrol/issues/6264

https://github.com/mavlink/qgroundcontrol/issues/6262

https://github.com/mavlink/qgroundcontrol/issues/6261

May be it is time to sort out the whole Joystick idea.

BTW the three mode 0,1 and 2 are pretty necessary if we have to use it with a pro ground station like this and simulations as well: 

[image]

We start to work now on a project that uses the PX4 only as a real simulation controller for X-Plane so the mode 2 will be pretty much needed.

Also now in mode 2 if you have connected the RC there is no way we could use it because the RC fights the Joystick all the time and the elevons are shaking.

In the ideal world we would love to have the RC and joystick connected at the same time and the craft accept commands from both depending on which one you touch. I know that this is a tough thing to do because there should be some priorities in the control but that will nail it. 

At the current stage our ground station Joystick is just an eye candy. :)

DonLakeFlyer, 2018-04-30:
So QGC+Joystick+SITL used to work out of the box. By default COM_RC_IN_MODE on SITL =  1. Yet QGC joysticks no longer work. I checked that joystick code is outputing values to MANUAL_CONTROL messages to the vehicle.

@dagar @LorenzMeier Something broken on firmware side. Could use some firmware folks to debug using QGC+Joystick.

dagar, 2018-05-01:
This is still working for me (current PX4 and QGC master as of 5 minutes ago). What's your setup?

DonLakeFlyer, 2018-05-01:
master of both tested with Logitech F710. Tried both: ```make posix gazebo_typhoon_h480``` and
```make posix gazebo_plane```

Strange

DonLakeFlyer, 2018-05-01:
Wait, I didn't notice it booted to Hold flight mode. When I arm in Stablized it tells me ```Failsafe enabled: no RC```

DonLakeFlyer, 2018-05-01:
How can I get an RC failsafe with COM_RC_IN_MODE=1 which is not supposed to be doing RC checks.

dagar, 2018-05-01:
RC checks actually refers to things like the throttle value at arming.

mhkabir, 2018-05-21:
This is blocking SITL testing of multiple PRs. Is there someone who knows how this broke?

dagar, 2018-05-21:
What is it blocking?

DonLakeFlyer, 2018-05-21:
I gave up trying to get it to work with the problems documented above still happening. I can try it again.

dagar, 2018-05-21:
This seems to be working for me with PX4 and QGC master as of about 15 minutes ago. I'm using Ubuntu with a Logitech gamepad and `make posix_sitl_default gazebo_iris`.

The only thing that was weird was defaulting "TX Mode" to 1 (left stick pitch instead of throttle), and for a gamepad having "Full down stick is zero throttle" works a lot better for altitude and position control modes.

[image]

[image]

Sidewinder1138, 2018-09-17:
Update: Mode 2 did end up working for me once I did the Radio setup/calibration and Flight Modes setup (didn't realize I needed to do that before).

TSC21, 2018-10-25:
@dagar @RomanBapst I can still reproduce this issue. I am trying to test the control surfaces (elevons) using the input of one of the channels but when I switch to Manual I get `Failsafe enabled: no RC` and I am unable to give any inputs to the control surfaces.

junwoo091400, 2022-11-22:
Not sure if this is still relevant?

junwoo091400, 2023-03-01:
Reminds me of this btw: https://discuss.px4.io/t/px4-community-q-a-february-22-2023/30831#q5-simulating-rc-inputs-josh-14