PX4/PX4-Autopilot #9320: Intel Aero RTF Indoor Handheld Tests Failing

jdiament95, 2018-04-17:
My Intel Aero RTF is equipped with a Lidar Lite v3 range finder, and I'm also using the optical flow algorithm running on the aero compute board. The PX4 firmware is using the EKF2 estimator with EKF2_AID_MASK=1 (optical flow) and EKF2_HGT_MODE=2 (range finder). 

I'm trying to get the drone to achieve a stable hover in position control mode. However, when performing handheld tests (no props and changing throttle), the state estimate occasionally goes wild (chaotic attitude and position estimates). Any idea why this could be happening?

dagar, 2018-04-17:
Could you share a log showing the issue? https://logs.px4.io/

Did you have this working previously?

mhkabir, 2018-04-20:
Probably this is because of the known issue with Lidar Lite 3 on Aero?

jdiament95, 2018-04-20:
Is there any way the spikes can be ignored by changing some parameter for the Kalman filter in the firmware?

Otherwise, is there another range finder that interfaces better with the aero? This documentation seems to suggest that the aero is the best option! (https://docs.px4.io/en/flight_controller/intel_aero.html)

hamishwillee, 2018-04-22:
> Is there any way the spikes can be ignored by changing some parameter for the Kalman filter in the firmware?

I don't know, but EKF docs are here: * https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html

jdiament95, 2018-04-23:
I've tried the tuning the parameters suggested on that page -- but altitude estimate is still diverging.

Also, using the MAVLink analyze widget, I observe that the distance sensor measurements are lagging behind actual variations in altitude by a few seconds. I tried incrementally increasing the EKF2_RNG_DELAY parameter from 5ms to 5000ms (which is beyond the suggested maximum 300ms) -- the altitude estimate is still diverging. I'm wondering if the lagging measurements of the lidar lite are more problematic than the spikes...

Is there anyone who has successfully used a Lidar Lite v3 on the Aero with PX4 firmware? @LorenzMeier

julianoes, 2019-07-10:
I'm not sure who could still support Intel Aero at this point, so I'm leaving this closed. Sorry about that.