PX4/PX4-Autopilot #9322: Intel AeroRTF: lidar lite v3 spikes in data

jdiament95, 2018-04-18:
I have a lidar lite v3 attaches via I2C to the compass port of the aero. I have an electrolytic capacitor attached between power and ground (as suggested in the lidar lite v3 manual). The compass port I2C connection is shared with the compass that is included on the aero. 

Viewing the raw range finder data via mavlink analyze widget shows recurring large spike spikes in the distance measurement. This seems to be a common issue with the aero [(see intel aero forum post)](https://communities.intel.com/thread/121441). The consensus on the aero forum is that the only way to avoid spikes is to read the data into an external microprocessor rather than the aero flight controller. Has anyone figured out a more feasible fix that doesn't require the extra link? I'm trying to fly the aero in altitude and position hold mode, so it would be much more convenient if the distance sensor data used by the autopilot firmware was free of spikes. Thanks!

mhkabir, 2018-04-18:
This can potentially help : https://github.com/PX4/Firmware/pull/9247

jdiament95, 2018-04-18:
I just upgraded to the latest firmware, and still getting lots of spikes with the distance sensor measurement.

`nsh> ver git` returns the following:

FW git-hash: 0f1c9504d57826a3bfacff0af214c9a3a7080af0
FW version: 1.7.4 0 (17236992)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: fe0dc6c41a6bd7d942945459ac960ff2d3e1f4fc

mhkabir, 2018-04-18:
@jdiament95 Updating the firmware won't help, there have been no driver improvements in a long time. Please try the patch above, or switch to a better sensor.

jdiament95, 2018-04-19:
I tried the patch above - the data still has large spikes.

Isn't the lidar lite v3 regarded as a fairly good sensor? I'm unsure why this seems to be such a recurring problem especially on the aero (see intel aero forum referenced in original post)...

mhkabir, 2018-04-19:
@zehortiga @anitha-suresh could you please help here?

zehortigoza, 2018-04-19:
I did some investigation and there is no much else to do to reduce the spikes than add the capacitor, it is probably some electrical problem.
My recommendation is change to a UART lidar like LeddarOne.

closed 2019-02-06 by stale[bot]