PX4/PX4-Autopilot #9325: Nxphlite Flight Controller Issues

RomanBapst, 2018-04-18:
1) Cannot upload mission, says 'unable to write to SD card'
- removing the 'dataman start' call from the init script and starting it manually from the terminal after boot solves the problem
- stopping and restarting dataman also solves the problem
- at the lowest level, the ::read/::write to the dataman file on the SD card fails, the errno is EBADF, which means 'Bad file number'

2) RC link seems flaky, sometimes does not connect, often needs reboot, also SBUS does not work

3) Telemetry link is awfully slow and sometimes QGC fails to load params
- possibly related to missing flow control

4) external mag is not detected as external, possibly because it's not started with -X option (different bus architecture)

5) sometimes board freezes on boot, needs reboot (can be seen if safety switch LED light does not blink)

dk7xe, 2018-04-18:
@RomanBapst please add
can not set 115200 baud as speed rate for TELEM2.
in case i try to set `3115200` via QGC on `SYS_COMPANION ` to get normal telemetry 115k2 bd i get 
`Value must be within 0 and 1921600`
 error message.
[image]

DanielePettenuzzo, 2018-04-19:
@dk7xe try to tick the force save box. This should allow you to set any value.

dagar, 2018-04-19:
The telem 2 settings (SYS_COMPANION) are enums and not directly used as baudrates. See the init script https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS#L626.

https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/system_params.c#L111-L140

I think we can do better than this, but it's a larger discussion. https://github.com/PX4/Firmware/issues/8302

RomanBapst, 2018-06-11:
@davids5 Thanks for the update!
@dk7xe Have you by chance already tested this and can you confirm that these issues are solved with the latest firmware?

davids5, 2018-07-01:
I think these are resolved on https://github.com/PX4/Firmware/tree/leitwert-nxphlite-serial-dma. Please reopen if not,

closed 2018-07-01 by davids5