PX4/PX4-Autopilot #9329: gamepad wishlist

dagar, 2018-04-18:
- throttle allow mapping full range [-1, 1] (cars, etc)
 - ideally VTOL transition should toggle from one state to the other
 - arm/disarm require at least 1 second

More advanced configuration
 - allow mapping throttle to other buttons as a +-% change, which frees up the stick for gimbal use

dagar, 2018-04-18:
@RomanBapst @MaEtUgR This obviously doesn't cover everything, but it's a few relatively easy short term fixes. Feel free to extend the list.

dagar, 2018-04-18:
- show thumbsticks in the Joystick tab
   - this is minor, but would give the user a better idea about what's going on and provide some necessary configuration (how the throttle is handled)

dagar, 2018-04-18:
This was intended to be a QGC issue, although there will also be a PX4 issue later for different pieces.
https://github.com/mavlink/qgroundcontrol/issues/6367

closed 2018-04-18 by dagar

MaEtUgR, 2018-04-26:
That went fast. It all sounds useful to me, I can probably work on the [-1,1] problem with the throttle (https://github.com/PX4/Firmware/issues/9331)