PX4/PX4-Autopilot #9332: QGC logging issues with Intel Aero RTF

jdiament95, 2018-04-19:
I'm following the instruction in [this tutorial](https://dev.px4.io/en/log/logging.html), but I can't consistently generate log files. Most of the time, when I click "start logging" no file appears under "Saved Log Files." Occasionally a file appears and the index begins increasing, but then it quickly disappears. 

Here's some info from the NuttX console:

`nsh> logger status
INFO  [logger] Running in mode: mavlink
INFO  [logger] Mavlink Logging Running`

`nsh> mavlink status
instance #0:
	mavlink chan: #0
	no telem status.
	flow control:	ON
	rates:
	tx: 0.512 kB/s
	txerr: 0.000 kB/s
	rx: 0.000 kB/s
	rate mult: 0.377
	accepting commands: YES, FTP enabled: NO
	transmitting enabled: YES
	mode: Normal
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS3 @57600)

instance #1:
	GCS heartbeat:	865846 us ago
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control:	ON
	rates:
	tx: 66.425 kB/s
	txerr: 0.028 kB/s
	rx: 0.805 kB/s
	rate mult: 1.000
	ULog rate: 40.0% of max 70.0%
	accepting commands: YES, FTP enabled: YES
	transmitting enabled: YES
	mode: Onboard
	MAVLink version: 2
	transport protocol: serial (/dev/ttyS1 @921600)`

bkueng, 2018-04-19:
Did you disable logging in mavlink-router? By default, logs are already stored on the linux side via mavlink router under `/var/lib/mavlink-router`.

jdiament95, 2018-04-19:
How could I check if I disabled logging in mavlink-router? I never intentionally did that.

Is the file path you mentioned on the aero compute board? If so, is there any way to get the logs to store on my laptop running QGC?

bkueng, 2018-04-19:
> How could I check if I disabled logging in mavlink-router?

There should be a line `Log=` in ` /etc/mavlink-router/main.conf` on the compute board that you can delete.

> Is the file path you mentioned on the aero compute board?

Yes. If you disable mavlink router ligging, you should be able to use the QGC log streaming, as you did above.

closed 2018-04-20 by jdiament95