PX4/PX4-Autopilot #9335: Nxphlite - UART4/TELEM2 output starts only when USB connected

dk7xe, 2018-04-19:
**Bug Report**

With PX4 version from April 12th the Mavlink datastream starts on UART4/TELEM2 only in case a second connection is made via USB.
TELEM2 / SYS_COMPANION is set to Normal Telemetry 57600 baud.
[image]

davids5, 2018-06-06:
@dk7xe - Please retest on https://github.com/PX4/Firmware/commits/leitwert-nxphlite-serial-dma 7fbca79724b9b4d41f4867c32493bdc3a6e7e043 

If you have a radio connected. Ensure you pull down Pin 5 on the radio and P10-4 on the FMU. Observe the LED if it is solid red the radio is in bootload mode.

davids5, 2018-07-02:
This was resolved with the serial and usb int storm fixes.

closed 2018-07-02 by davids5