PX4/PX4-Autopilot #9338: submodule lib/ecl unable to checkout

potaito, 2018-04-19:
I am unable to checkout the most recent commit of `lib/ecl` https://github.com/PX4/ecl/commit/8678cc42ba6c87b3ae0724af2becb62b71bf6683.

When freshly cloning the px4 firmware and then updating the submodules, I also get the error:
```
$ git clone git@github.com:PX4/Firmware.git
$ cd Firmware
$ git submodule init
$ git submodule update --recursive
```

I then end up with this:
```
$ git status

On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

	modified:   src/lib/ecl (new commits, modified content)

no changes added to commit (use "git add" and/or "git commit -a")
```

Not sure where the problem is or how this slipped through CI :neutral_face:

potaito, 2018-04-19:
Will push fix in a moment

dagar, 2018-04-19:
This sometimes happens when a Firmware change with corresponding submodule change goes in. For the sake of testing and review the Firmware is pointed at another branch in the submodule that it requires. They're both merged (and branches deleted), and now the exact submodule commit is missing.

Example: https://github.com/PX4/Firmware/pull/9319

We need to make sure the submodule commits in Firmware master/beta/stable always point at submodule commit that's going to exist indefinitely.

potaito, 2018-04-19:
Got it. I think I fixed it by checking out the corresponding commit that exists on libecl's master: 
https://github.com/PX4/Firmware/pull/9339

PR #9339 (2018-04-19): libecl: checked out recent commit of master
Fixes #9338 

Currently the submodule `src/lib/ecl` points to this commit, which is not on master: https://github.com/PX4/ecl/tree/8678cc42ba6c87b3ae0724af2becb62b71bf6683

The commit is on master, but under a different commit hash: https://github.com/PX4/ecl/commit/bd72f3c5211176351cbc0596ef69395d359be43d

What I did for fixing it: 
```
cd src/lib/ecl
git checkout bd72f3c5211176351cbc0596ef69395d359be43d
cd ..
git commit
```

@dagar can you verify?
closed 2018-04-19 by dagar