PX4/PX4-Autopilot #9343: Feature Request/Discussion: return to ground control station (RT_GCS)

Stifael, 2018-04-19:
There is an easy way to do that, but probably not the best one. 

Px4/QGC already supports follow-me mode, which uses number #144 as mavlink message.  However, the logic in the mc_pos_controller_main.cpp is already broken to some extend (for instance. the velocity demand [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1421-L1442) will never pass the velocity controller). 

Therefore, it makes sense to fix it and maybe even extend it such that the same message can be used for the RT_GCS feature. 
I was discussing this feature with @dogmaphobic who mentioned that further modification might be required:
>the code would need to check if it's using a high latency link (Iridium satellite) for instance and control the frequency of updates to a significant lower rate. It would also need UI asking the user if this is to be enabled or not and the desired frequency

Any suggestions are welcomed.

dagar, 2018-04-19:
Could you describe high level how you'd expect this to function keeping in mind the existing HOME position and RTL behaviour?

For example would you simply want to have the home position automatically updated to the location of the ground control station if it's available? Keeping in mind the ramifications for failsafes and mission items with relative altitudes. 

Alternatively we could handle it as the actual ground control station location. Then "return to ground station" and "return to home" would be distinct things.

I would initially try to think this through separately from the existing follow-me mode, although some of the pieces could probably be shared later on.

Antiheavy, 2018-04-19:
I think this is a neat idea and probably would have use with certain types of precision landing systems and other applications.

As the number of  different "Return" functions increase, it would be nice to keep an organized framework, if possible.  I don't contribute code, but I'll contribute my thoughts from a user perspective as separate entries below.

Antiheavy, 2018-04-19:
There is already some messages and a (psuedo?) mode type referred to as "Return to Ground station" or "RTGS".  I believe it triggers in certain loss of RC situations.
https://github.com/PX4/Firmware/blob/0caa4dc1712771faec3ca57756998dedfd29bea7/msg/vehicle_status.msg#L22

The terminology, descriptions, code comments, and functionality is wrong may be somewhat broken.  What occurs when this triggers is basically a Return to Home type functionality.  It shows up as "Return to Ground station" in the Qgroundcontrol UI. 

If a true "return to ground station" function was implemented, this old and wrong code should get cleaned up and removed to avoid confusion.

Antiheavy, 2018-04-19:
In my mind this is what the Return framework looks like:
[image]

Antiheavy, 2018-04-19:
Terminology can get out of control (already is).  for example:

acronyms used: RTL, RTH, RTGS, RT_GCS
terminology used: Return to Land, Return to Launch, Return to Home, Return to Groundstation, Return to Ground Control Station, Smart RTL, Mission RTL, etc...

I find it useful to think of just one "Return" flight mode.  "Return" mode can have a number of parameter selectable behaviors.  I can't tell you what variable names to use in your code, but from a user perspective things should be kept simple and organized with just one "Return" flight mode and user selectable sub behaviors.

There is a param called "RTL_LAND_TYPE" (which might soon change to "RTL_TYPE").  This is one method to handle the various sub behaviors.

hamishwillee, 2018-04-19:
@Antiheavy FYI current documentation is setting on only [Return](https://docs.px4.io/en/flight_modes/return.html) - though  sometimes with aliases - e.g. Return mode (a.k.a. Return to launch - RTL, Return to home - RTH) etc). I completely agree with your sentiments above. This should also integrate in a user friendly way with precision landing.

I LOVE your diagram [above](https://github.com/PX4/Firmware/issues/9343#issuecomment-382906714). Once this round of work is done on return, lets get a modified version of that into the docs. Do you have a shareable link that we can use for editing when that happens (I like docs to have a link so we can edit things later).

junwoo091400, 2022-11-22:
Still relevant? @Stifael