PX4/PX4-Autopilot #9346: Virtual Joystick issue in HITL Simulation

barzanisar, 2018-04-20:
Issue 1) I am using Pixracer in HITL. I followed instructions on https://dev.px4.io/en/simulation/hitl.html?q  to perform HITL simulation. I wanted to use virtual joystick, which I enabled from Application Settings > General and set COM_RC_IN_MODE parameter to "virtual RC by joystick"

However when I fly in any mode requiring RC input, the vehicle keeps yawing even though my joysticks are centered as shown below:
[image]

Issue 2) When I move the left joystick up/down, the vehicle yaws, whereas moving left joystick right/left , vehicle pitches. For right joystick, up/down is throttle and left/right is roll. This is not the same as normal mode 2 joysticks/ RCs. How can we change this?

I tried both Gazebo and Jmavsim and I am using the latest master of Firmware.

dagar, 2018-04-20:
Do you mean physical (usb) joystick? QGC on screen thumbsticks are also sent over mavlink.

Related - https://github.com/PX4/Firmware/issues/9318

barzanisar, 2018-04-20:
@dagar I mean QGC on screen thumbsticks

thomasgubler, 2018-04-20:
I think this might be a QGC issue? @dogmaphobic

closed 2019-02-06 by stale[bot]