PX4/PX4-Autopilot #9349: Offboard mode (Position triplet setpoint) ignores specified coordinate frame

szebedy, 2018-04-21:
As already posted in [#5407](https://github.com/PX4/Firmware/issues/5407), it is not possible to create a position setpoint for the drone in body frame (e.g. fly 1m forward, 1m to left), only in local NED coordinates. For more details check out [issue #5407](https://github.com/PX4/Firmware/issues/5407). However I tested and it works for velocity setpoints instead of position setpoints.

I believe the implementation of the feature (position setpoint in body coordinates (FRAME_BODY_OFFSET_NED)) should not require so much effort, but would be very useful for vision based control applications.

If you guys are all swamped can anybody at least suggest a workaround? Something like transforming a target point from body coordinates to local NED coordinates with ROS/tf.

dagar, 2018-04-21:
A relatively quick fix would be implementing it in the SET_POSITION_TARGET_LOCAL_NED Mavlink receiver (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L836) rather than in the controller itself.

szebedy, 2018-04-21:
@dagar Thank you so much for the instant reply. Can you assign somebody to do this or are you suggesting that I could implement it?

dagar, 2018-04-21:
This is something where it's probably not that hard to implement once you have the pieces together to test it. If you'd like to implement it feel free to ping me and I'll try to point you in the right direction.

szebedy, 2018-04-22:
@dagar I would be happy to implement it, however some pointers would be very helpful. I took a look at the code, and I saw that the coordinate frame is accessible in the received message in set_position_target_local_ned.coordinate_frame.

As I mentioned earlier, the body frame works for velocity control, and [here you can see](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L948) that the coordinate frame is given to the pos_sp_triplet struct, however only at velocity control. I'm guessing the first step would be doing the same thing at position control, then conditioning on the frame at a later point. What do you think?

dagar, 2018-04-22:
Somewhere you need to take the offset and apply it to the current local position.

You could either do that in the mavlink_receiver (close to where you linked), or in the MC position controller  https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L734. 

The first problem that comes to mind is that OFFBOARD has a failsafe/timeout if the messages aren't sent continuously.

closed 2019-02-06 by stale[bot]