PX4/PX4-Autopilot #9353: Unable to start mission ('vehicle not ready')

sticks918, 2018-04-22:
Hi all, I've been manually flying a Pixracer for a while and finally hooked up a GPS to try autonomous missions. Despite a solid GPS lock and a simple takeoff -> land plan uploaded, I can't start any missions.

It seems to arm without error, but when starting any mission the error I get is 'Unable to start mission: vehicle not ready'

I've uploaded the log here: https://logs.px4.io/plot_app?log=4b12bb08-a51a-48f2-b8b7-df5cf1c8cf81
though it's pretty short so I'm not sure how helpful it is.

Can anyone help me figure out what's causing it to not start any missions? Thanks!

dagar, 2018-04-22:
Did you see any other errors in QGroundControl? The GPS fix was not good enough for a valid position estimate (what the flight controller actually uses).

[image]

Was this inside? 

There's ongoing work to improve this. https://github.com/mavlink/qgroundcontrol/pull/6364

dagar, 2018-04-23:
There are commander eph/epv thresholds for initializing a home position, and the EKF has its own thresholds (EKF2_REQ_EPH, etc).

The vehicle not ready message you saw is on the QGC side. If the attempt made is as far as the vehicle it would have probably told you that the transition to MISSION mode is denied due to lack of valid global position.

In the PX4 log (.ulg) there's a message logged called vehicle_status_flags which is what the system is ultimately using. The important one in this case is condition_global_position_valid.

sticks918, 2018-04-23:
Ok I'll try again with after letting it sit in an open area for a little longer. Is there a way to check those flags (vehicle_status_flags and condition_global_position_valid) live in QGC or is it only available through the log post-flight?

dagar, 2018-04-23:
Once the mavlink message `GLOBAL_POSITION_INT` starts streaming (see mavlink inspector) that means the estimator is publishing a valid global position estimate (good gps below thresholds).

sticks918, 2018-04-23:
Ok I'll do some more digging. Thanks

closed 2018-04-23 by sticks918

acpl00, 2018-12-14:
I have QGC-Station and fly-mission-plan, when i set up all the points and I load the mission, I get the error 'Unable to start mission ('vehicle not ready')'.

In other words,

I have tried position control and the drone responded quite satisfactorily using the RC transmitter. However, when I want to try fly mission mode, I can see that the GPS position is jumping even though the QGC takes up till 9 satellites, and the GPS is locked and the HDOP < 1.5 and the VDOP < 1.5.

I can chose the way points and load the program but when I want to change to mission and arm I get the error: I get the error 'Unable to start mission ('vehicle not ready')'.