PX4/PX4-Autopilot #9355: EKF no global position

JohnSnowball, 2018-04-23:
**Bug Report** 
I'm now using a RTK gps of **5Hz**. During flight, I found in some time dead reckoning flag is true and vel_pos_innovation becomes extremly large which leads to no global position failsafe. 
I'm not sure why it happens, I checked GPS signal, it's almost stable of 5Hz, is it due to gps lat\lon jump?
Or do I need to modify EKF2_GPS_DELAY param to avoid this? Does EKF2 support 5Hz GPS signal? 
Here is the log:
https://logs.px4.io/plot_app?log=bcbd64f5-8549-404f-b338-30b03152ee6a

Please also check this figure shown below: gps.eph is normal, gps fix is also the best(fixed), but vel_pos_innov[3] and vel_pos_innov[4] is abnormally high, thus global_position.eph is bigger than threshold:
[image]

dagar, 2018-04-23:
How old is your branch? I ask because I noticed the EKF2_REC_RPL parameter was removed a long time ago, and I recall adjusting things like the GPS delay likely around the same time.

Most of the commonly used GPS modules (ublox m8n) are actually 5Hz. 

Could you try again with PX4 master?

dagar, 2018-04-23:
@priseborough any recommendations for using EKF2 with an RTK GPS? Worth setting GPS antenna positions perhaps?