PX4/PX4-Autopilot #9373: IRlock driver in v1.7.4 Beta

zquang, 2018-04-27:
I want to use precland mode with IRlock camera. It seems that precland is supported on v1.7.4 Beta, not on v1.7.3 Stable.
In v1.7.4 Beta, if I type "irlock start", "irlock test" and show up the IR marker (MarkOne Beacon V2.0), then the FW shows some detection results (3-10 times) and abruptly RESET the Pixhawk, stop all of motors if it is arming (tested in both case of arming and disarming).

But if I switch back to v1.7.3 Stable, the irlock driver works fine, just don't have "precland" mode.

mhkabir, 2018-05-02:
@ndepal Could you have a look?

ndepal, 2018-05-02:
We are using the irlock driver on a PixRacer and  have never seen these issues. But I also never use `irlock test`, perhaps that's what causes problems.

Does it only crash if you do `irlock test`? What happens if you do `irlock start` and then `listener irlock_report` once you hold the beacon in front of the camera?

What board are you using?

Using a PixRacer with PX4 1.7.4 I can't reproduce this issue.

zquang, 2018-05-02:
I am using  Pixhawk 2.4.8 board and enable driver/irlock module in v2-default.cmake file
I cannot fit listener module in the flash (just 1MB) for your suggestion. But I modify px4_simple_app to listen to irlock_report topic. It works without any problem when I show the IR marker infront of the IR camera.
So I think the problem comes from "irlock test" function.

closed 2018-05-02 by zquang