PX4/PX4-Autopilot #9375: Yaw rate command in Offboard mode through set_attitude_target message

mandelyoni, 2018-04-27:
**Bug Report** 

Dear Sir,

We're running in Offboard mode and sending a set_attitude_target message through Mavlink.
The PX4 version is 1.6.5 on Intel Aero.

It seems that the roll and pitch rate commands work, ok but the yaw rate doesn't work at all. This is the code from the Mavlink_receiver.cpp   :

MavlinkReceiver::handle_message_set_attitude_target(mavlink_message_t *msg) {
...
....
....
rates_sp.roll = set_attitude_target.body_roll_rate;
rates_sp.pitch = set_attitude_target.body_pitch_rate;
rates_sp.yaw = set_attitude_target.body_yaw_rate;

}

Please advise.

Thanks, 

For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](slack.px4.io). Issue reports need to contain the items below:

 - Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/flight-reporting.html))
 - Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/)
 - Expected behavior and actual behavior.
 - Steps to reproduce the problem.
 - Specifications like the version of the project, operating system, or hardware.

TSC21, 2018-04-27:
I advise you to update the Firmware version. A lot of changes has come since 1.6.5. Use lastest stable or master.
Also, without knowing how are you actually sending those commands and what code is generating those, it's almost impossible to help you debug this.

mandelyoni, 2018-04-27:
**The commands are being sent through the following part of the next function:** 

MavlinkReceiver::handle_message_set_attitude_target(mavlink_message_t *msg)  {
....
                ///XXX add support for ignoring individual axes
                if (!(_offboard_control_mode.ignore_bodyrate)) {
                    vehicle_rates_setpoint_s rates_sp = {};
                    rates_sp.timestamp = hrt_absolute_time();

                    if (!ignore_bodyrate_msg) { // only copy att rates sp if message contained new data
                        rates_sp.roll = set_attitude_target.body_roll_rate;
                        rates_sp.pitch = set_attitude_target.body_pitch_rate;
                        rates_sp.yaw = set_attitude_target.body_yaw_rate;
                    }

                    if (!_offboard_control_mode.ignore_thrust) { // dont't overwrite thrust if it's invalid
                        rates_sp.thrust = set_attitude_target.thrust;
                    }

                    if (_rates_sp_pub == nullptr) {
                        _rates_sp_pub = orb_advertise(ORB_ID(vehicle_rates_setpoint), &rates_sp);

                    } else {
                        orb_publish(ORB_ID(vehicle_rates_setpoint), _rates_sp_pub, &rates_sp);
                    }

TSC21, 2018-04-27:
Well I am not getting what are you trying to say. The above code is found [here](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1218). And that's the code you use on the Firmware side to process the `SET_ATTITUDE_TARGET` MAVLink messages. But for processing those msgs, something else on the other end needs to send those msgs.

TSC21, 2018-04-27:
This was fixed some days ago - https://github.com/mavlink/mavros/pull/753. Please use MAVROS master while a new release is not done.

closed 2018-04-27 by TSC21

mandelyoni, 2018-04-28:
Dear TSC21,  

please see the following error we received, while installing Mavros master according to  - https://github.com/mavlink/mavros/tree/master/mavros.  Any idea?  Thank you.

yoni@yoni:~/catkin_ws$ rosdep install --from-paths src --ignore-src -y
ERROR: Rosdep experienced an error: Package version ":{version}" does not follow version conventions
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.12.1

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 140, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 388, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 443, in _package_args_handler
    pkgs = find_catkin_packages_in(path, options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py", line 33, in find_catkin_packages_in
    packages = find_packages(path)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 86, in find_packages
    packages = find_packages_allowing_duplicates(basepath, exclude_paths=exclude_paths, exclude_subspaces=exclude_subspaces, warnings=warnings)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 153, in find_packages_allowing_duplicates
    path_parsed_packages, warnings_lists = zip(*pool.map(parser, data))
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 251, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 567, in get
    raise self._value
InvalidPackage: Package version ":{version}" does not follow version conventions

TSC21, 2018-04-28:
@mandelyoni that is not related with the Firmware anymore. Please move this discussion to MAVROS gitter.

mandelyoni, 2018-04-29:
Hi,

I gave the same set_attitude_target command using mavlink (not ROS) but I still can't control the yaw rate (the mode is OFFBOARD).
Similar to Ros, I can control through Mavlink the roll rate and pitch rate but not the yaw rate (or the yaw).
Is there any other Mavlink command I can use (in Offboard mode) to command the yaw rate or yaw?

Thank you

ashraf-badr, 2021-06-26:
@mandelyoni 
Hello form the other side :D
I have the same issue. I am using PX4 in offboard mode to control the drone through mavlink. 
It accepts the roll and pitch but never give ant response to the yaw commands.
Did you solve this issue?
Any help will be appreciated.