PX4/PX4-Autopilot #9379: PX4 / QGC Sync on Tuning Sliders

LorenzMeier, 2018-04-28:
The current tuning sliders in QGC link to parameters that we recently disabled. We need to map them to the new correct attitude parameters. @barzanisar I believe that @thomasgubler did discuss this with you last week. Have you filed issues for that or derived specs with @RomanBapst for it?

@barzanisar Please do this yourself as you already have the build setup for QGC. Here are the three tuning panels:
https://github.com/mavlink/qgroundcontrol/blob/16541062efa0acee5e128f61dcd23106a7afc1f6/src/AutoPilotPlugins/PX4/PX4TuningComponent.cc#L55-L78

Here is e.g. the one for Multicopters:
https://github.com/mavlink/qgroundcontrol/blob/16541062efa0acee5e128f61dcd23106a7afc1f6/src/AutoPilotPlugins/PX4/PX4TuningComponentCopter.qml

@RomanBapst Please take responsibility for making sure that all parameters and the wording make sense. @MaEtUgR @Stifael it would be great if you would contribute to this.

barzanisar, 2018-04-29:
@LorenzMeier I don't remember discussing this issue with @thomasgubler but I can look into it this week.

barzanisar, 2018-04-30:
As discussed with @RomanBapst, we should first discuss which parameters need to be exposed on the Tuning Setup. Things we discussed:

- Parameters for "Hover Throttle" and "Manual minimum throttle" exist and are correctly mapped.
- Sliders for "Roll sensitivity" and "Pitch sensitivity" should either map to "MC_ROLL_P and MC_PITCH_P" or "MC_ROLLRATE_P and MC_PITCHRATE_P".
- In my experience I have tuned MC_ROLLRATE_P more than MC_ROLL_P to achieve either a faster roll control response or reduce oscillations. So maybe we can consider mapping Rate parameters to the "roll/pitch sensitivity" sliders?

I will also try tuning s500 with just MC_ROLL/PITCH_P and see if they are useful enough to be exposed as tuning sliders.

barzanisar, 2018-05-09:
@DonLakeFlyer @dogmaphobic 
[image]

When I compile master or [stable v3.3](https://github.com/mavlink/qgroundcontrol/tree/Stable_V3.3) branch, I get these weird versions of tuning sliders. However, when I open the binary version, the sliders look normal. Can anyone figure out why this is so? Is there any special way of compiling to see the tuning sliders?

@RomanBapst the fix for this issue is in my [fork](https://github.com/barzanisar/qgroundcontrol/blob/pr-barza-tuning-sliders/src/AutoPilotPlugins/PX4/PX4TuningComponentCopter.qml#L65-L207). Can you please confirm the values? 
Also, I checked the tuning sliders for [Plane](https://github.com/barzanisar/qgroundcontrol/blob/pr-barza-tuning-sliders/src/AutoPilotPlugins/PX4/PX4TuningComponentPlane.qml#L29-L62) and [VTOL](https://github.com/barzanisar/qgroundcontrol/blob/pr-barza-tuning-sliders/src/AutoPilotPlugins/PX4/PX4TuningComponentVTOL.qml#L29-L81). They all exist. Do we need any changes there? Do we need to add any new parameters?

DonLakeFlyer, 2018-05-09:
Can you move any QGC problems to QGC Issues?

taileron, 2018-05-14:
often I´m using the iOS version of QGC which is not above v 3.2 and it looks for old tuning parameters like mc_roll_tc or mc_pitch_tc that are no more existing in current PX4 firmwares.
So the mapping should not be only handled by QCG otherwise newer PX4 with older QCG don´t match anymore. PX4 firmware should be able to translate tuning parameters into older QGC as well

thomasgubler, 2018-05-18:
Superseded by mavlink/qgroundcontrol#6438

closed 2018-05-18 by thomasgubler