PX4/PX4-Autopilot #9381: Should QGC add support for DO_SET_HOME in missions?

DonLakeFlyer, 2018-04-29:
Discussion started here: https://github.com/mavlink/qgroundcontrol/issues/6371#issuecomment-385265600

Antiheavy, 2018-04-30:
This needs to be thought through VERY carefully as HOME setpoint is safety critical and impacts many things.

Concepts of HOME location and HOME (field) altitude need to be considered and how changing either of those impacts things during, before, or after flight. 

The idea that HOME gets updated upon takeoff (or arming) is a fairly universal behavior in the small UAV world these days (always exceptions, but...).

dagar, 2018-04-30:
Within commander we now have better support for setting this manually and not auto updating once it's been set manually. The missing piece is the interaction via mavlink or within a mission.

When is a planned home position needed or wanted?

Personally I'd be happy if QGC only ever displayed the real PX4 home position.

DonLakeFlyer, 2018-04-30:
This has nothing to do with "Planned home position". This is if you explicity add a DO_SET_HOME command somewhere in your mission.

> Personally I'd be happy if QGC only ever displayed the real PX4 home position.

I think people that plan missions ahead of time will not be happy with that. It means no mission time calculation, no battery count calculation, ... amoung other things.

dagar, 2018-04-30:
> This has nothing to do with "Planned home position". This is if you explicity add a DO_SET_HOME command somewhere in your mission.

I'm saying it could be if you had QGC send DO_SET_HOME when a user manually set it.
Again, how do people actually want to use this?

> I think people that plan missions ahead of time will not be happy with that. It means no mission time calculation, no battery count calculation, ... amoung other things.

I suspect some of those users think placing the home position is actually doing something on the vehicle.

dagar, 2018-04-30:
Could we make it entirely optional? The default for QGC + PX4 could be not showing the home position unless it's streaming from the vehicle. Then a user could opt into placing a home position for the sake of planning and qgc stats.

Later if anyone actually wants a manually set home position we can plumb it together easily enough.

DonLakeFlyer, 2018-04-30:
> The default for QGC + PX4 could be not showing the home position unless it's streaming from the vehicle.

That pretty much means you will never get correct stat calcs (time, batteries, bottom alt/terrain display, ...) until you are at the field and the vehicle is in the position you plan to take off from. Which is about as close to never as you can get. That is not super useful.

barzanisar, 2018-05-03:
I personally think "Planned Home" should be replaced with "Virtual Home". To me, the word "Planned" means the home position I planned while planning the mission, which could also lead into confusion with DO_SET_HOME position (once that command is added). On the other hand, "Virtual Home" gives a more strong feeling that this home is not real.

barzanisar, 2018-05-03:
> I think people that plan missions ahead of time will not be happy with that. It means no mission time calculation, no battery count calculation, ... amoung other things.

Where does the new QGC show mission times and battery count calculation?

hamishwillee, 2018-05-04:
> I get what you are saying about Planned. Not sure about Virtual. Maybe Expected would be better.

FWIW

My preference would be "Placeholder Home". I would be OK with: Planning, Planned, Nominal, Notional. Either way, it should be visibly different.

Virtual is too overloaded as a term - raises more questions than it answers. No so sure for Expected.

closed 2019-02-06 by stale[bot]