PX4/PX4-Autopilot #9389: [load_mon] init low on stack!(268 byte left)

wkyoun, 2018-05-01:
@dagar  @mhkabir 

PX4 Quadrotor | Open 3D View
-- | --

irframe: | DJI Flame Wheel F450Quadrotor x (4011)
-- | --
Hardware: | PX4FMU_V2
Software Version: | 305859f3branch: master
OS Version: | NuttX, v7.22.0
Estimator: | EKF2

https://review.px4.io/plot_app?log=a24cc6ed-4da0-488e-b9de-6194c273dfdd

Would you please let me know what caused the this following warning message and how to fix it?

[image]

bkueng, 2018-05-01:
Do you see this repeatedly or just after changing the airframe/configuration?

BTW your setup is extremely noisy (https://review.px4.io/plot_app?log=a24cc6ed-4da0-488e-b9de-6194c273dfdd#Nav-Raw-Acceleration), I'm surprised that this flies.

wkyoun, 2018-05-01:
@bkueng

Yes, I was trying to implement your commit for second GPS(telem 2)

The following log was the log for the same vehicle.

https://logs.px4.io/plot_app?log=3027489e-bfee-44b4-9dff-97ed295723b3

https://review.px4.io/plot_app?log=9366072f-4c3c-4764-a423-c3af96eed63b

https://review.px4.io/plot_app?log=a33c93f2-b05a-48a4-9cac-4ea56a1688ce

bkueng, 2018-05-02:
Do you have other local changes?

You can set [CONFIG_USERMAIN_STACKSIZE](https://github.com/PX4/Firmware/blob/master/platforms/nuttx/nuttx-configs/px4fmu-v2/nsh/defconfig#L836) to 2600, then use `make clean` and flash again.

wkyoun, 2018-05-03:
@bkueng 

Yes, I have implemented local changes.

Do this following warning message influence other module performances?(EKF2 or Control)
```[load_mon] init low on stack!(268 byte left)```

bkueng, 2018-05-03:
> Do this following warning message influence other module performances?

No, not at all, it's only a warning.

v01d, 2018-07-11:
I'm getting this same warning periodically.

tstellanova, 2018-09-06:
I'm consistently getting this with Pixhawk 4 running  v1.8.0, on maybe 25% of reboots. It appears that it is `init` that is running low on stack: 

`top` without low stack warning:
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   3 init                        11439  0.000  2120/ 2484 100 (100)  w:sem  3
```
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   3 init                        12569  0.000  2168/ 2484 100 (100)  w:sem  3
```

`top` with low stack warning:
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   3 init                        12874  0.000  2216/ 2484 100 (100)  w:sem  3
```
This corresponds to the console warning:
`WARN  [load_mon] init low on stack! (268 bytes left)`

I am not changing anything in e.g. init scripts before rebooting.

tstellanova, 2018-09-06:
Increasing init's stack (`CONFIG_USERMAIN_STACKSIZE`) seems to alleviate the problem, similar to https://github.com/PX4/Firmware/commit/d7580aa676b502b755281a0c7a7e5780b1df9066

closed 2019-02-06 by stale[bot]

bigbellmercy, 2019-03-02:
In my case, this 'init low on stack!' message happened after changing the airframe from the generic quad x to DJI F330 in PX4 v1.8.2 on Pixhawk2.1.

bigbellmercy, 2019-03-20:
> Do you have other local changes?
> 
> You can set [CONFIG_USERMAIN_STACKSIZE](https://github.com/PX4/Firmware/blob/master/platforms/nuttx/nuttx-configs/px4fmu-v2/nsh/defconfig#L836) to 2600, then use `make clean` and flash again.

The link above does not exist anymore, so, instead, use the link below:
[CONFIG_USERMAIN_STACKSIZE](https://github.com/PX4/Firmware/blob/f13bbacd5277123d6af2cb5ed21587c220031353/platforms/nuttx/nuttx-configs/px4fmu-v2/nsh/defconfig#L836)

dagar, 2019-03-20:
This should be resolved on master.