PX4/PX4-Autopilot #9397: why there is no value of position setpoint triplet data at the beginning

EVANSONG1220, 2018-05-02:
I am using PX4 on the multicopter in offboard mode. On one flight, I wanted the yaw angle holded at its initial angle, about -102 degree, but it didn't follow my intention and turned to the 0 degree, after about 10 seconds, it came back to its initial angle(-102). I examined the ulg data and found that  there was no value of position setpoint triplet data and the vehicle attitude setpoint.yaw_body was 0 which should be the multicopter's actual initial yaw angle.  The part of data plot on the following picture, maybe help to understand my problem.
[image]
THIS IS THE DATA FILE:
[data.zip](https://github.com/PX4/Firmware/files/1966471/data.zip)

dagar, 2018-05-04:
Which message were you sending?

This might be the timing with initialization.

EVANSONG1220, 2018-05-07:
@dagar Thank you for your response. I was sending the horizontal velocity, vertical velocity and yaw angle rate which I wanted the drone to follow. I used the same date and same way to control for several times, but encountered this situation only this time. How can I know when the initialization is completed? which data can show me this progress?

closed 2019-02-06 by stale[bot]