PX4/PX4-Autopilot #9398: Flow not working anymore with default parameters

ChristophTobler, 2018-05-02:
https://github.com/PX4/Firmware/commit/dc2d6e8aab4148f4672e3cf674c7e85f86cf5533 breaks the use of optical flow with default parameters. I had to change `COM_POS_FS_EPH` to about `20` for https://github.com/PX4/Firmware/commit/dc2d6e8aab4148f4672e3cf674c7e85f86cf5533#diff-1d51a813dcb26f46616f71f28aacc70dR3753 to work (EKF2 reports an EPH of about 14 when using flow in simulation).

Can we avoid having one fixed `EPH` value?

dagar, 2018-05-02:
The plan was to expand the failsafe to have different thresholds based on the context. However, even before that PR commander still only had one eph value. How was this working before?

Do you have a semi-recent log from before that change?

priseborough, 2018-05-07:
This is fixed with the following ecl patch: https://github.com/PX4/ecl/commit/25e98be359ef477d2f1d30d42b833926f42d9fbc 

However we should fix it in the commander because position uncertainty grows over time when reliant on optical flow and we don't want an undetermined time in flight for the position accuracy check to fail. See fix here: https://github.com/YUNEEC/Firmware/commit/1a7847886079cdcebbb2f9ab9dd63b81f777839d

These are both part of a number of optical flow use improvements in https://github.com/YUNEEC/Firmware/tree/tap-v3-product-optFlowFixes-wip

priseborough, 2018-05-07:
I will rebase the ecl changes and raise a PR.

dagar, 2018-05-07:
@ChristophTobler could you describe the full use case? Then I can update the state machine and failsafe handling appropriately.