PX4/PX4-Autopilot #9402: MAVLink set_home_position in local frame coordinate system

alexx600, 2018-05-03:
Hello,
I would like to ask if it is possible to change home position using local NED x,y not global lattitude, longitude coordinate?

According to http://mavlink.org/messages/common#SET_HOME_POSITION “The global and local positions encode the position in the respective coordinate frames”.
What shoud i send as a lattitude and longitute if i would like to move home position in local frame (for example 2 m forwad)?
I tried set lattituede and longitude to zero, but this caused reseting home position. Is it possible to change home position in local system?

dagar, 2018-05-03:
Could you explain the desired use case? Currently all internal consumers of the home position only use the global position, although I'd like to see this expanded. 

This is where the command is handled. https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L847

alexx600, 2018-05-04:
'm doing a small part of a bigger project. The landing platform is to be located on the car - it is movable. The drone has a camera with an additional single board computer, which is responsible for recognizing the landing site and its relative position. The vision system does not have a GPS module. Communication between the image recognition system and the pixhawk is one-sided. It is not possible to read the GPS location from the autopilot.

The image recognition system sends the SET_TARGET_POSITION_NED commands. It would be good to update the home position so that in the event of an accident (RTL) it could land on the landing platform and not 200 m further.

TSC21, 2018-05-04:
@alexx600 RTL currently depends on LLA coordinates. You would have to implement a specific case for RTL that would only rely on local coordinates.

alexx600, 2018-05-04:
Maybe (probably) i am wrong but if pixhawk get home position from GPS during start, and later i send move home forward, it  should recalc home position to lla coordinate. If GPS coordinate is known, heading and z,y local is known IT should be posibble to recalc local movement to lla

TSC21, 2018-05-04:
> It is not possible to read the GPS location from the autopilot.

I read this above. So if it is not possible, how are you feeding the position to the vehicle? From the vision system? If yes, then you are right we have a way of reprojecting into the global coordinates (though we are removing this in the feature). But still, the current RTL implementation relies on the LLA coordinates and not in the local position. You would have to extend it, besides extending setting the home position with local coordinates.

xdwgood, 2018-05-07:
@alexx600  Do you need a dynamic RTL?  :smiley:

lamping7, 2018-05-07:
Why not just do a frame transformation?

alexx600, 2018-05-07:
> why would you want to set an home position to a position that changes fast with time (a mobile platform)?

Car speed is quite small, drone ought to be tethered and be used as a artificial observation point. I'm not a decision-maker. Reckognition system is in develop now. I would like to have an emergency stop that i could press from rc tranmisstor.  In that case emergency stop, should be understood as a langing procedure trigger. Landing on platform or on the floor few meters behind car.

> I would advise you to look into to explore the possibility of using precision landing mode as a "failsafe" option instead of RTL.

Thank you for advise. In the coordinator's opinion, we must have RTL. Special during vision system tests.

alexx600, 2018-05-07:
@xdwgood I realize that it may sound distant but yes.

alexx600, 2018-05-07:
@lamping7 As i mentioned above Vision system do not have direct access to pixhawk GPS module.  reckognition system does't have geographic copter coordinate only relative landing platform location.

I always can use raspberry pi zero as a companion computer, It could read GPS data from pixahwk and relative langing platform position from vision module. It would have all nesesery data to landing platform coordinate transform, and home position change.  But i would say that method is not very elegant.

lamping7, 2018-05-07:
Can you not simply just land in place? I'm not sure of your environment.

closed 2019-02-06 by stale[bot]