PX4/PX4-Autopilot #9409: ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.

rahulsharma11, 2018-05-04:
Hi, I have just upgrade to Ubuntu18. From then whenever i build the px4 firmware i got the error-
ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Is this really due to Ubuntu Upgradation? How can i solve it?
Thanks

dagar, 2018-05-04:
Yes this is likely Ubuntu updated related.

What are you running exactly? PX4 itself shouldn't be using gtk3, is that from gazebo?

rahulsharma11, 2018-05-04:
Hi, Its not from gazebo. I am building px4fmu-v2_default.

dagar, 2018-05-04:
Exactly when do you see the error? Screenshot?

dagar, 2018-05-04:
Weird. Did you do a fresh install of Ubuntu 18.04 or an upgrade from an older release?

rahulsharma11, 2018-05-04:
I upgraded from previous 17.10

rahulsharma11, 2018-05-07:
Hi,
Any suggestion?

benVolatiles, 2018-05-09:
I get the same error after updating to Ubuntu 18.04 when building our firmware (completely unrelated to PX4) too, so I'd say this is a bug in Ubuntu.

The firmware build is completely unaffected by this nuisance.

rahulsharma11, 2018-05-09:
Ok. Thanks for the confirmation.

closed 2018-05-09 by rahulsharma11

imambungo, 2018-08-26:
I solved the same issue by adding the following line in `.bashrc` file in my home folder:

    export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libgtk3-nocsd.so.0

Make sure you have `libgtk3-nocsd.so.0` in `/usr/lib/x86_64-linux-gnu/`
If you didn't have it, try `sudo apt-get install gtk3-nocsd`

worstprogrammerCN, 2018-10-15:
> I solved the same issue by adding the following line in `.bashrc` file in my home folder:
> 
> ```
> export LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libgtk3-nocsd.so.0
> ```
> Make sure you have `libgtk3-nocsd.so.0` in `/usr/lib/x86_64-linux-gnu/`
> If you didn't have it, try `sudo apt-get install gtk3-nocsd`

That works for me. Besides, you can run ```find /usr -name libgtk3-nocsd.so.0 -print``` to find it.

imambungo, 2018-10-15:
Happy to help :blush:

neokos90, 2018-11-10:
I had the same problem and this solve helped me too. Thanks!

wizardion, 2019-02-13:
It is helped with me too! Thanks!

SPlatten, 2019-05-27:
I was running Ubuntu 16.04 in a virtual machine on iMAC, I stupidly updated to 18.04 and now I'm getting these issues, unfortunately none of the fixes I've found so far have helped and I don't have the folder /usr/lib/x86_64-linux-gnu, could this be because I'm on an iMAC ?

Bost, 2019-07-23:
See also https://github.com/lutris/lutris/issues/905#issuecomment-514185518

Syed-Mohammad-Nasir-Hussain, 2021-06-09:
There are lot of .bashrc files,which one has to be selected and moreover when trying to edit and save file its saying you dont have access for writing the file