PX4/PX4-Autopilot #9410: rc override :  command an Intel Aero drone through /mavros/rc/override topic and receive no response

mandelyoni, 2018-05-04:
Hi,

I'm trying to command an Intel Aero drone through /mavros/rc/override topic and receive no response.
Any ideas? link to code samples?

Thank you,  Yoni.

dagar, 2018-05-04:
Do you have regular RC (spektrum) connected as well?

Sharing the flight log might also help. https://logs.px4.io/

mandelyoni, 2018-05-05:
Yes, a remote controller is connected.
Is it supposed to work?
Here   ,   https://github.com/mavlink/mavros/issues/22 ,    MHKABIR noted that:   "This won't work. We don't handle RC override in PX4 stack."  

Thank you

lamping7, 2018-05-07:
I've used it about 1 - 1.5 yrs ago as a hack to actuate a payload release. It did work, but the behavior was kind of odd and took some experimentation. I had to repeatedly command high, it took a few messages from mavros to "go", while checking the feedback pwm topic to see it actuated, upon which I would stop publishing, and it then would resume to the state of whatever the RC was commanding (sorry for the run-on sentence). FYI, This method is generally not advised.

closed 2019-02-06 by stale[bot]