PX4/PX4-Autopilot #9428: RTPS Serial Port Initialisation Insufficient

drisebor, 2018-05-07:
The msg/templates/urtps/microRTPS_transport.cpp file only sets up one of the termios configuration flags for the serial port, ie in UART_node::init() :

// Clear ONLCR flag (which appends a CR for every LF)
uart_config.c_oflag &= ~ONLCR;

This causes unpredictable behaviour on some platforms depending on the previous serial port setup.

When running posix_sitl_rtps on Ubuntu, accessing the same serial port beforehand as used for RTPS using PuTTY but with different settings causes serial communications to fail (no bytes received). For example setting up odd parity using putty prevents any communication over RTPS. Communication may also fail immediately after Ubuntu boots. 

The setup code needs to be modified to lock down all relevant flags and variables in termios to non-canonical, binary transmit and receive, 8 bits, 1 stop bit, no parity, which is what RTPS over serial on the Pixhawk currently defaults to.

bkueng, 2018-05-08:
I agree. Can you send a PR please?

drisebor, 2018-05-08:
PR done. It should be ok. It's a very comprehensive lockdown of the Termios configuration.

TSC21, 2018-12-10:
Solved.

closed 2018-12-10 by TSC21