PX4/PX4-Autopilot #9432: Start logging on RTK GPS condition or postprocessing option

thomasgubler, 2018-05-07:
**Feature Request**

From @priseborough 

> Be able to conditionally start the data log when the RTK GPS goes into a specified mode. This will prevent excessive length log files for replay becasue getting the GPS into full RTK becasue getting the GPS into full RTK mode can take 10 minutes or more.

@bkueng Another option I see is to add a post processing option to split log files based on the RTK GPS mode. Please discuss with @priseborough directly what makes more sense for development/tuning of RTK.

The goal is that we are able to fine tune RTK to normal GPS handover (see related tickets).

Related:
* https://github.com/PX4/Firmware/issues/9431
* https://github.com/PX4/Firmware/issues/9422

AuterionWrikeBot, 2018-05-31:
➤ Beat Küng commented:

Talked to Paul, this is probably not needed

priseborough, 2018-05-31:
Yes, I am told that provided the rover is not powered up until the base station has finished its survey, the time for the rover GPS to get an RTK fix is not long.

closed 2018-05-31 by priseborough