PX4/PX4-Autopilot #9439: SITL multi-vehicle data link issue

Ahrovan, 2018-05-08:
Alternatively problems occur 
```
INFO  [commander] data link #0 lost
INFO  [commander] data link #0 regained
```
and can not control vehicles truly.
 - ubuntu 16.04 - ros kinetic
 - based on [multi-vehicle-simulation](https://dev.px4.io/en/simulation/multi-vehicle-simulation.html)
 - System info:
Processor	Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz, 4001 Mhz, 4 Core(s), 8 Logical Processor(s)
Physical Memory (RAM)	24.0 GB
NVIDIA GeForce GTX 980 Ti

```
roslaunch px4 multi_uav_mavros_sitl.launch gui:=false... logging to /home/cv/.ros/log/97fd1b0a-528b-11e8-adbf-cb24266246e5/roslaunch-ubuntu-11904.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:46209/

SUMMARY
========

CLEAR PARAMETERS
 * /uav1/mavros/
 * /uav2/mavros/
 * /uav3/mavros/
 * /uav4/mavros/
 * /uav5/mavros/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /uav1/mavros/cmd/use_comp_id_system_control: False
 * /uav1/mavros/conn/heartbeat_rate: 1.0
 * /uav1/mavros/conn/system_time_rate: 1.0
 * /uav1/mavros/conn/timeout: 10.0
 * /uav1/mavros/conn/timesync_rate: 10.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /uav1/mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /uav1/mavros/distance_sensor/laser_1_sub/id: 3
 * /uav1/mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/laser_1_sub/subscriber: True
 * /uav1/mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /uav1/mavros/distance_sensor/lidarlite_pub/id: 1
 * /uav1/mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /uav1/mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /uav1/mavros/distance_sensor/sonar_1_sub/id: 2
 * /uav1/mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /uav1/mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /uav1/mavros/fake_gps/eph: 2.0
 * /uav1/mavros/fake_gps/epv: 2.0
 * /uav1/mavros/fake_gps/fix_type: 3
 * /uav1/mavros/fake_gps/geo_origin/alt: 408.0
 * /uav1/mavros/fake_gps/geo_origin/lat: 47.3667
 * /uav1/mavros/fake_gps/geo_origin/lon: 8.55
 * /uav1/mavros/fake_gps/gps_rate: 5.0
 * /uav1/mavros/fake_gps/mocap_transform: True
 * /uav1/mavros/fake_gps/satellites_visible: 5
[... 1284 lines cut ...]
WARN  [commander] MANUAL CONTROL LOST (at t=22161ms)
WARN  [commander] MANUAL CONTROL LOST (at t=23339ms)
INFO  [commander] data link #0 lost
pxh> ~INFO  [commander] data link #0 regained
WARN  [commander] MANUAL CONTROL LOST (at t=26457ms)
pxh> ~WARN  [commander] MANUAL CONTROL LOST (at t=28583ms)
WARN  [commander] MANUAL CONTROL LOST (at t=30471ms)
WARN  [commander] MANUAL CONTROL LOST (at t=31174ms)
INFO  [commander] data link #0 lost
WARN  [commander] MANUAL CONTROL LOST (at t=32836ms)
INFO  [commander] data link #0 regained
WARN  [commander] MANUAL CONTROL LOST (at t=34493ms)
WARN  [commander] MANUAL CONTROL LOST (at t=38044ms)
INFO  [commander] data link #0 lost
INFO  [commander] data link #0 regained
INFO  [commander] data link #0 lost
WARN  [commander] MANUAL CONTROL LOST (at t=41600ms)
WARN  [commander] MANUAL CONTROL LOST (at t=43974ms)
WARN  [commander] MANUAL CONTROL LOST (at t=44915ms)
INFO  [commander] data link #0 lost
WARN  [commander] MANUAL CONTROL LOST (at t=47266ms)
WARN  [commander] MANUAL CONTROL LOST (at t=49618ms)
INFO  [commander] data link #0 regained
WARN  [commander] MANUAL CONTROL LOST (at t=50321ms)
```

lamping7, 2018-05-10:
Hard to say. Port issue? Also note all the sensor failures. How does this work with less vehicles? 2? 3?

Ahrovan, 2018-05-13:
I installed a new os and problem solved. please close this issue

closed 2018-05-13 by lamping7