PX4/PX4-Autopilot #9446: GPS driver debugging best method

trgiman, 2018-05-10:
Hi. I would like to better understand GPS driver section in the PX4 code. Especially ashtech driver, that I am modifying for "indoor gps module". Using gdb would be the best option. I need to debug step by step usage of buffers in the code. But while in the SITL mode GPS driver is not used actually. 

And since I am using Aero RTF I do not know if  it would be the option to connect STLinkv2 for debugging? 

I must be missing something fundamental. Thanks for suggestions.

trgiman, 2018-05-11:
@bkueng Thanks. It means that in the case of gps driver in SITL. -d can be for instance "GPS" unit connected to /dev/ttyUSB0 (in case of Ubuntu). Correct?

trgiman, 2018-05-12:
Hi @bkueng. It works as you suggested. While I am in the pxh shell, listener command for vehicle_gps_location topic returns the indoor GPS values (since I already modified ashtech driver a bit in rgrds of NMEA readings). But when I start the gdb session and set some brake point for ashtechdriver.ccp. It seems I can not "arrive" to break point since gdb for step or next commands reports messages as for example following:

> Thread 8 "wkr_hrt" received signal SIGINT, Interrupt.
> [Switching to Thread 0x7ffff5543700 (LWP 4239)]
> 0x00007ffff702930d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
> 84	../sysdeps/unix/syscall-template.S: No such file or directory.

SITL simulation without gdb is otherwise working as expected as described above. Any suggestions?

LorenzMeier, 2018-05-12:
You need to disable interrupt signals in GDB. Nothing is wrong with this, it just stops by default on signals.

trgiman, 2018-05-12:
Hi @LorenzMeier thank you for guidance. I just handled Thread 4 "hpwork" received signal SIGCONT, Continued and it works as needed.

closed 2018-05-12 by trgiman