PX4/PX4-Autopilot #9447: PX4 SITL catkin build failing

jbourne15, 2018-05-10:
I am following the instructions from here https://dev.px4.io/en/setup/dev_env_linux.html#gazebo-with-ros but when the script gets to catkin build it throws an error.  I don't know if this is relevant but I had already installed this successfully on the same computer, but after deleting the files and reinstalling I am getting that the catkin build fails.

Here is my log file.
[build.make.000.log](https://github.com/PX4/Firmware/files/1992424/build.make.000.log)

thanks for the help

TSC21, 2018-05-10:
You need to update your submodules. Try `git submodule update --init --recursive`. Then `catkin clean --all -y` and `catkin build` it again.

jbourne15, 2018-05-11:
Still getting the same catkin build error (from the src/Firmware directory).

log file:
[build.make.001.log](https://github.com/PX4/Firmware/files/1993794/build.make.001.log)

dagar, 2018-05-13:
I believe this is fixed ecl side. https://github.com/PX4/ecl/pull/447

closed 2018-05-18 by dagar

jbourne15, 2018-06-19:
@dagar I am installing this again and I am getting the following errors:

> fatal error: ecl.h: No such file or directory compilation terminated.

> fatal error: geo/geo.h: No such file or directory

Log file:
[build.make.002.log](https://github.com/PX4/Firmware/files/2115358/build.make.002.log)

I have made sure that I have the latest firmware  with the [change](https://github.com/PX4/ecl/pull/447) you mentioned earlier and have ran `git submodule update --init --recursive` and `catkin clean --all -y`.  Thanks for any help.

dagar, 2018-06-20:
@jbourne15 I don't know why catkin cmake isn't respecting the variable (path to ecl.h). Can you move over to https://github.com/PX4/Firmware/pull/9679 and continue the discussion?

commit ac4053fb (2018-05-18): Update submodule ecl to latest Fri May 18 01:26:47 UTC 2018 (#9484)

 * cb63f16 2018-05-17 Daniel Agar - cmake cache ECL source directory path to work with catkin (#447)
 * fixes #9447